chipsalliance/UHDM #59: Headers don't include what they use

hzeller, 2020-01-14:
Looking at a random header, e.g. `headers/variables.h`: it does not include any headers, yet it uses `SymbolFactory::getSymbol()` for instance.

So the header only works if wherever it is used the header  `vpi_uhdm.h` that declares this symbol happens to be included beforehand.

Each include file must include all the dependencies it uses, including <vector>, <string> etc.

alainmarcel, 2020-01-15:
uhdm.h is the only header needed when using this UHDM.
Users don't have to figure out header dependencies, uhdm.h contains all the sorted dependencies.
Do you think this is still worth changing?

hzeller, 2020-01-15:
Unless something are 'internal headers', we should make them self-contained. Headers that work by accident depending on the sequence they are included is problematic.

Each class should include all the elements it actually needs. If we have pointers to other objects it _might_ be ok to just forward declare them, but I'd also say the relevant headers should be included.

Having 'super headers' that include everything in a particular order makes code hard to navigate and understand, makes it hard to determine dependencies or do refactorings.

So yes, even if we have a `uhdm.h` convenience header, each include file should be self-contained and include what they use.

hzeller, 2020-01-15:
Another example of living dangerously: the *.cpp files 'strategically' typedef a type before including other headers. Here an example from 
`templates/Serializer_restore.cpp`

```c++
#include <vector>
#include <map>
typedef void any;
#include "headers/containers.h"
//...
```

This is dangerous. If we need an 'any' type that is typedeffed, we should put it in a header and include where we need it.

In general, all the headers in a *.cpp file shoud be organized in any order (they order they should be included is: c-headers, c++ headers, projecte header; in each of these blocks sorted alphabetically).

https://google.github.io/styleguide/cppguide.html#Self_contained_Headers

hzeller, 2020-01-15:
An example how fragile things are currently: if you
```
clang-format -i src/vpi_user.cpp
```
which arranges headers, the resulting file is not compilable anymore

hzeller, 2021-07-16:
Testing including separate files, this results still in issues.
I'll be sending a few pull requests to address this.

hs-apotell, 2022-02-10:
#635 resolves this issue for all headers. Some implementation files are still using `uhdm.h` but I supposed these files will end up needing all model files anyways.

PR #90 (2020-01-22): Cleanup headers
Fixes #59
closed 2022-08-23 by alaindargelas