chipsalliance/UHDM #63: Can the #defienes be an enumeration ?

hzeller, 2020-01-14:
Given that the C++ part allows us to use all kinds of features we can use there, I wonder if we should replace the
  #define ... list in `headers/uhdm.h` with a C++ `enum class`.

#defines are very annoying as they pollute the global namespace.

alainmarcel, 2020-01-15:
There are 1000s of define in vpi_user.h and sv_vpi_user.h, I only add a few more.
We can't get rid of the IEEE ones, is it worth getting rid of these? They kinda work hand-in-hand

hzeller, 2020-01-15:
So I guess if this is something we only do within our own implementation, we should be strict and use the strictest way to define things these days.

If we expect the end-user actually using these values as 'extension' to whatever vpi-values exist, then we should maybe prefix them with `uhdm_` ?

hzeller, 2020-01-15:
Oh, I see, they are already prefixed with `uhdm`. But still they might be more readable with an additional underscore `uhdm_` to indicate that that is essentially the 'namespace'.

PR #89 (2020-01-21): Use an enum for uhdm types
Fixes #63
closed 2020-01-21 by alainmarcel