chipsalliance/UHDM #73: Generate dot file

hzeller, 2020-01-15:
Low prio.

Nice to have: since we know all the dependencies, we can also generate a dot-file to create a gigantic graph using [graphviz](https://www.graphviz.org/). We can use that when visualizing or explaining things.

Typically straight-forward to generate (emit all the nodes and edges, the rest happens automatically), but not clear how useful it would be though, large graphs tend to be ... large.

mithro, 2020-01-15:
Doxygen could generate this potentially from the C++ code.

Nic30, 2020-05-26:
Do you mean the dot for class diagram or the dot for nodes described using UHDM? For visualization of hardware it is much better to use ELK as it can generate much better layered graphs.

e.g. I am using it in [d3-hwschematic](https://github.com/Nic30/d3-hwschematic)

hzeller, 2020-05-26:
Interesting, I didn't know about ELK. That looks indeed very nice!

Nic30, 2020-05-26:
[adaptagrams](https://github.com/mjwybrow/adaptagrams) can do the same thing and it may better for your project as it is C++.

hzeller, 2020-07-06:
First experiment: directly generating a graphiz output of the hierarchy as described in the yaml files without too much tweaking does not result in an overly useful output yet.
(I've left out the `BaseClass` to reduce arrow-clutter, but then this resulted in the ones directly extending BaseClass to show up on the right of the graph. Anyway, just an visualization experiment at this point)

[image]

hzeller, 2021-09-17:
I created another dot file with the current model; Color coded `obj` is blue, `class` green and groups red. Since rendering groups as separate nodes will make it more messy, I just printed all the groups a class or object is member of in the red box.

The amount of objects and the way graphviz auto-renders the edges makes it look a bit messy; ideally the edges would just do be rendered with a 'bar connect' 
[image]
Anyway, some tool that is more specicifally suited for this hierachical representation should be chosen. Haven't looked at the tool @Nic30 proposed above yet.

Anyway, here is a PNG of that. To see any details, you'd have to open it in a separate tab possibly.

Easier to look at (but github doesn't allow to upload as attachment, hence zipped): below attached two versions of this and a horizontal version (nicer to read, but requires more horizontal scrolling) as SVG.
[image]

[left-right.svg.gz](https://github.com/chipsalliance/UHDM/files/7187284/left-right.svg.gz)
[top-bottom.svg.gz](https://github.com/chipsalliance/UHDM/files/7187287/top-bottom.svg.gz)