chipsalliance/UHDM #77: Run clang-format automatically on everything

mithro, 2020-01-15:
All code should have clang-format run on it. Travis should run clang-format and check that the format is correct.

alainmarcel, 2020-01-22:
This is not applicable here. Not at checkin time at least. Maybe as a compilation step part of the Makefile?
All the source code that we check in is template code, we do not check in the actual generated code.
clang-format does not know how to deal with meta/template code.

hzeller, 2021-09-08:
While the templates won't work, we should clang-format at least the final include files that we install.

This will actually find some subtle missing IWYU issues, as it sorts the headers alphabetically and will expose things that have not been included before; I've previously spent some time to make it work, but seems to be broken again.

I'll take this one and look to run clang-format in cmake as postprocessing of the generated code.