chipsalliance/UHDM #176: Add kokoro users as admins of this repo

mithro, 2020-04-14:
- `kokoro-team` user -- Admin permissions
 - `symbiflow-kokoro` user -- Admin permissions
 - `symbiflow-robot` user -- Triage permissions

This enables Kokoro to setup hooks.

alaindargelas, 2020-04-14:
@kokoro-team
kokoro-team
Awaiting kokoro-team’s response
Pending Invite
 
@symbiflow-kokoro
SymbiFlow Robot (Kokoro Runner)
Awaiting symbiflow-kokoro’s response
Pending Invite
 
@symbiflow-robot
SymbiFlow Robot
Awaiting symbiflow-robot’s response

closed 2020-04-14 by alaindargelas

mithro, 2020-04-14:
@symbiflow-kokoro and @symbiflow-robot should have accepted.

alaindargelas, 2020-04-15:
Tim,
I had to convert the account alainmarcel to be an organization to allow
these.
I think I somehow lost the travis aut build in the process, any idea?

alaindargelas, 2020-04-15:
It works now after I synced the accounts.