chipsalliance/UHDM #207: Generate a class hierarchy diagram for generated models

hs-apotell, 2020-05-07:
Title says it all. Annotating the class whether or not it is abstract will be helpful.

alaindargelas, 2020-05-07:
BTW, this information is in the Verilog DataModel itself,
this file only contains the DataModel diagrams for easy search, abstract classes are dotted-line:
https://github.com/alainmarcel/UHDM/blob/master/third_party/Verilog_Object_Model.pdf

I find this documentation much better than any automatically generated diagram.

Also, in the existing headers/*.h, the Object Factory is #if 0 #endif for the Abstract classes (You can't make an object).

What form of annotation would you prefer if the above does not suffice?

hs-apotell, 2020-05-08:
> I find this documentation much better than any automatically generated diagram

The documentation you are pointing to is a good read but not from a quick reference point of view right in the middle of programming session serving as a reference guide. I contend that there is a need for a supporting document.

I find myself asking this question quite often - _I need access to function ```VpiName``` (or some other function), what is the nearest class that I can typecast a pointer to so that I get the information without having to cast it to the most derived class._ This question cannot be answered by just browsing the corresponding header (```UHDM::expr``` doesn't tell me that ```UHDM::ref_obj``` subclass' it)

I am still familiarizing myself with the source code. I am sure this pain will ease up with time as I memorize the hierarchy. But I would argue that anyone would have similar questions while onboarding.

Here's a good example from [Qt](https://doc.qt.io/qt-5/hierarchy.html) and [Java](https://docs.oracle.com/javase/8/docs/api/overview-tree.html)

My primary interest is the class hierarchy which will be immediately useful. The annotation calling out specific class whether or not it is abstract is a wish item.

hzeller, 2021-09-17:
I once played aroud with generating dot files in #73 , but the graphviz rendering is not entirely useful and looks a bit messy due to the way each edge is rendered.

hs-apotell, 2021-09-17:
Check out plantuml and elk layout. Python also has a few graph rendering engines that might be worth checking out.