chipsalliance/UHDM #210: Add support for Windows and Mac

hs-apotell, 2020-05-08:
Title says it all.

hs-apotell, 2020-05-08:
I have made some progress on this and hit a wall.

- I fixed up model_gen.tcl to work on windows.
- Major blocker - Use of ```process``` and ```forever``` are undocumented reserved words when building with VC compiler.
- Major blocker - Use of ```final``` as class name. I am not even sure why and how this even builds successfully with gcc and clang. ```final``` is a C++ reserved word.

One potential solution would be to convert the class names to use Pascal casing rather than old school C-style naming convention being employed. Arguably this will be a big change but would conform with wider array of compilers and platforms.

Appreciate some guidance on what's an acceptable solution for these issues.

alaindargelas, 2020-05-08:
That is an easy fix, I'll rename the classes.
Note that model_gen.tcl is getting changed these days to support the elaborator
I'll let you know when I'm done with these changes.

alaindargelas, 2020-05-09:
I renamed those 3 classes. You should be unblocked to proceed.

hs-apotell, 2020-05-13:
@alaindargelas I have UHDM building and running on Windows. However, it's crashing while restoring saved UHDM file. I am not familiar with the archiving format yet and it would take me longer than what I can commit immediately. Have you got time to look into the crash?

alaindargelas, 2020-05-13:
Yes I can, dont merge, You mean it crashes on Linux now?Can you share a branch?

hs-apotell, 2020-05-13:
It’s crashing only on windows. I already have a pull request for the changes that should allow you to successfully build and reproduce the crash.

alaindargelas, 2020-05-13:
Ah, then no I cannot help there. I don't even have a machine to run it on Windows, and I have 0 experience debugging on Windows.Can you send a stack trace of the crash in debug mode?

hs-apotell, 2020-05-13:
Here's the callstack and also the uhdm file that its attempting to restore. Briefly looking at the serialization code, it looks like the archives are platform independent. You might be able to try restoring the file on Linux for debugging the issue. I noticed that the saved file is slightly bigger on windows (11 bytes) compared to what it generates on Linux. May be the issue is not restore and its the save operation.

[surelog.txt](https://github.com/alainmarcel/UHDM/files/4624341/surelog.txt) * Change the file extension to "uhdm"

```
uhdm-test2.exe!_CxxThrowException(void * pExceptionObject=0x0000008f4d1a8600, const _s__ThrowInfo * pThrowInfo=0x00007ff65cd998c8) Line 75
uhdm-test2.exe!kj::ExceptionCallback::RootExceptionCallback::onFatalException(kj::Exception && exception={...}) Line 921
uhdm-test2.exe!kj::throwFatalException(kj::Exception && exception={...}, unsigned int ignoreCount=2) Line 980
uhdm-test2.exe!kj::_::Debug::Fault::fatal() Line 343
uhdm-test2.exe!kj::BufferedInputStream::getReadBuffer() Line 117
uhdm-test2.exe!capnp::_::PackedInputStream::tryRead(void * dst=0x000001fd42d174a0, unsigned __int64 minBytes=7224, unsigned __int64 maxBytes=7224) Line 89
uhdm-test2.exe!kj::InputStream::read(void * buffer=0x000001fd42d174a0, unsigned __int64 minBytes=7224, unsigned __int64 maxBytes=7224) Line 52
uhdm-test2.exe!kj::InputStream::read(void * buffer=0x000001fd42d174a0, unsigned __int64 bytes=7224) Line 61
uhdm-test2.exe!capnp::InputStreamMessageReader::InputStreamMessageReader(kj::InputStream & inputStream={...}, capnp::ReaderOptions options={...}, kj::ArrayPtr<capnp::word> scratchSpace={...}) Line 225
uhdm-test2.exe!capnp::PackedMessageReader::PackedMessageReader(kj::BufferedInputStream & inputStream={...}, capnp::ReaderOptions options={...}, kj::ArrayPtr<capnp::word> scratchSpace={...}) Line 438
uhdm-test2.exe!capnp::PackedFdMessageReader::PackedFdMessageReader(int fd=3, capnp::ReaderOptions options={...}, kj::ArrayPtr<capnp::word> scratchSpace={...}) Line 447
uhdm-test2.exe!UHDM::Serializer::Restore(std::string file={...}) Line 61
uhdm-test2.exe!main(int argc=1, char * * argv=0x000001fd42d06910) Line 18
uhdm-test2.exe!invoke_main() Line 79
```

alaindargelas, 2020-05-13:
I see, it is in the buffer loading part of capnp.
Can you try to run this test first on Windows:

Surelog/dbuild/dist/Debug$ ./uhdm-test1

And trim the input design (In the .cpp) to be bare minimum?
Surelog/third_party/UHDM/tests/test1.cpp if it still fail, until there is
no object at all.

In this single test, the writer and loader are invoked from the same
executable, so we can rule out different version of capnp used for writing
and loading (which I think is happening).

hs-apotell, 2020-05-13:
test1 fails with the same callstack. I was attempting to debug the serialization and was using test2 at that moment, so I shared that callstack. I will try your suggestion about stripping the design to bare minimum and see if that helps narrow down the root cause.

hs-apotell, 2020-05-15:
Windows builds are now supported!

closed 2020-05-15 by hs-apotell