chipsalliance/UHDM #239: Missing column information in parsed data

hs-apotell, 2020-05-19:
UHDM::BaseClass provides access to File name and Line No. but the column number is not getting propagated to it from the parser.

Food for thought - Would it be worth (or at least an option to) carrying forward the actual token data i.e. the ANTLR primitive forward to allow access to other critical information for the purposes of linting. Being able to tie the model back to actual token can be useful.

hs-apotell, 2020-05-25:
We would need little more data actually - we need start/end line and column number. This information is available in Antlr AST node.

> What other property do you think is missing from the AST for Linting? Maybe the Column info is the only one?

Never say never! Our requirements today are just the start/end line/column number but we are so early in our development that I can't envision all the potential scenarios. If it's easy enough to make it available, I would recommend doing it - makes it a one stop solution. If it's not, then we will evaluate our future need when that time comes.

alaindargelas, 2020-05-26:
Alright, since this language-level linting does not require persistence, unlike design-level analysis, having the Antlr AST live data is probably the best way.
I'll pass it to UHDM via the UserData payload so it's optional and not intrusive in UHDM. It will be specific to Surelog front-end and populated optionally. I'll describe the mechanism here when this is ready.

hs-apotell, 2020-06-18:
Any update on this request?

alaindargelas, 2020-06-18:
No time to work on this for now. I need to get the full Earlgrey design supported. Might take another month or so.I can guide you if you want to add the support for columns. We would need even begin and end for each token. Antlr supports it.

hs-apotell, 2020-06-19:
Sure I can pick that up. Will ping back once I am done with the task at hand.

alaindargelas, 2020-08-25:
I gave more thoughts about this ask.
It is quite a daunting task. There are 2 possibilities:
- Only add the column start and maybe column end for the object (I'm still not sure about the end part, reading Antlr 
documentation).
   Strong Pro: Independant of Antlr, other parsers beside Surelog can fill those fields
   Pro: persistant on disk
   Weak Cons: other data member of the Token class not exported

- Providing a pointer to the antlr::token or antlr::ParserRuleContext in UHDM using the UHDM::BaseClass::ClientData pointer as a holder.
  Pro: all Antlr meta data about the tokens is accessible
  Cons: That client data attached to UHDM nodes will only be present with Surelog

No matter which solution, it requires to massively edit every place in Surelog where UHDM objects gets created to fill the new info, plus Surelog needs to actually carry over either the new column begin-end or the Antlr pointer in a lot of its internal datastructures.

The lesser of the 2 evil is to carry over the column being (and end) in my opinion.

hs-apotell, 2020-08-26:
@alaindargelas I don't think using the BaseClass::ClientData is a good idea. That is intended to be used for library users and not by the library itself. If anything BaseClass should hold an additional pointer to a generic type which could be filled in by the model code generator (tcl script).

Another potential solution is to not add it to UHDM at all. Implement it at the level of Surelog where access to ANTLR wouldn't be a problem. It does have the disadvantage of being completely untied to the modeled component hierarchy and potential additional bookkeeping to ensure proper cleanup. Elaboration might need special attention since the information need to be carried forward into the new hierarchy.

You bring up an interesting question - How does persistence work with this information? We do need this information to be persistent across loads without having to do the heavy lifting of reparsing the files. AFAIK ANTLR isn't persistent (and even if it is it might not be in the same way as UHDM does it). Given this requirement, I think there is only one solution.

Thoughts?

alaindargelas, 2020-08-26:
@Henner Zeller also suggested that I create an object that contains the information the Antlr Token contains (at least the one we are interested in for now) and make that an object of the UHDM data model (persisted). No matter what front-end implementation, users will interact with that facade object.

hs-apotell, 2020-08-27:
@alaindargelas @hzeller That's a reasonable solution. That will also ensure that adding new information doesn't change the model itself. Might help keeping things backward compatible.

hs-apotell, 2021-03-13:
Can we move this task to Surelog repository? Also, is this task on the near future roadmap?

alaindargelas, 2021-03-14:
Yes, please do. I think we should get to this before it reaches a year anniversary!
Let me finish the support for BlackParrot, then I'll switch to this.