chipsalliance/UHDM #267: How much memory does building UHDM require?

mithro, 2020-06-20:
The current Kokoro build on master failed with 
```
virtual memory exhausted: Cannot allocate memory
c++: internal compiler error: Segmentation fault (program cc1plus)
```

The computer has 4 cores and 16gb of memory;
```
========================================
Host CPU
----------------------------------------
Cores: 4

Memory
----------------------------------------
MemTotal:       15400288 kB
```

Results can be found at https://source.cloud.google.com/results/invocations/1be3f980-13d5-4a03-8ac8-115187c30862/log
```
[ 80%] Building CXX object CMakeFiles/uhdm.dir/src/variable_loads_group.cpp.o
[ 82%] Building CXX object CMakeFiles/uhdm.dir/src/variable_drivers_group.cpp.o
virtual memory exhausted: Cannot allocate memory
virtual memory exhausted: Cannot allocate memory
virtual memory exhausted: Cannot allocate memory
virtual memory exhausted: Cannot allocate memory
virtual memory exhausted: Cannot allocate memory
c++: internal compiler error: Segmentation fault (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
CMakeFiles/uhdm.dir/build.make:352: recipe for target 'CMakeFiles/uhdm.dir/src/vpi_user.cpp.o' failed
make[3]: *** [CMakeFiles/uhdm.dir/src/vpi_user.cpp.o] Error 1
make[3]: *** Waiting for unfinished jobs....
CMakeFiles/uhdm.dir/build.make:365: recipe for target 'CMakeFiles/uhdm.dir/src/SymbolFactory.cpp.o' failed
make[3]: *** [CMakeFiles/uhdm.dir/src/SymbolFactory.cpp.o] Error 1
CMakeFiles/uhdm.dir/build.make:378: recipe for target 'CMakeFiles/uhdm.dir/src/Serializer_save.cpp.o' failed
make[3]: *** [CMakeFiles/uhdm.dir/src/Serializer_save.cpp.o] Error 1
CMakeFiles/uhdm.dir/build.make:391: recipe for target 'CMakeFiles/uhdm.dir/src/Serializer_restore.cpp.o' failed
make[3]: *** [CMakeFiles/uhdm.dir/src/Serializer_restore.cpp.o] Error 1
CMakeFiles/uhdm.dir/build.make:417: recipe for target 'CMakeFiles/uhdm.dir/src/clone_tree.cpp.o' failed
make[3]: *** [CMakeFiles/uhdm.dir/src/clone_tree.cpp.o] Error 1
CMakeFiles/uhdm.dir/build.make:469: recipe for target 'CMakeFiles/uhdm.dir/src/ref_obj_interf_net_var_group.cpp.o' failed
make[3]: *** [CMakeFiles/uhdm.dir/src/ref_obj_interf_net_var_group.cpp.o] Error 4
make[3]: Leaving directory '/tmpfs/src/github/alainmarcel-uhdm/build'
CMakeFiles/Makefile2:350: recipe for target 'CMakeFiles/uhdm.dir/all' failed
make[2]: *** [CMakeFiles/uhdm.dir/all] Error 2
make[2]: Leaving directory '/tmpfs/src/github/alainmarcel-uhdm/build'
Makefile:160: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/tmpfs/src/github/alainmarcel-uhdm/build'
Makefile:5: recipe for target 'release' failed
make: *** [release] Error 2

[ID: 1696955] Build finished after 262 secs, exit value: 2

Warning: Permanently added 'localhost' (ECDSA) to the list of known hosts.
[13:20:08] Collecting build artifacts from build VM
Build script failed with exit code: 2
```

mithro, 2020-06-20:
I reran the Kokoro builder and it passed this time. I'm not sure what is happening here?

alaindargelas, 2020-06-22:
The largest file compiles using 1.5Gb in gcc.

closed 2020-06-23 by alaindargelas