chipsalliance/UHDM #289: SymbolFactory: SymbolID: should it start with 1 instead of 0 ?

hzeller, 2020-06-30:
In code like this `VpiFullName()` implementation https://github.com/alainmarcel/UHDM/blob/master/model_gen/model_gen.tcl#L125
we use an assumption that the Symbol-ID is 0 when not initialized, but in the symbol factory, the first valid ID is actually 0 ( https://github.com/alainmarcel/UHDM/blob/master/templates/SymbolFactory.h#L59 ).

This looks like a bug to me unless some symbol is explicitly inserted at the start of a symbol factory. If not, I think we should start with 1 to have the convenient *zero means uninitialized* condition assumed in the `VpiFullName()`

hzeller, 2020-06-30:
(I can take care of it, if it is confirmed an actual issue)

alaindargelas, 2020-07-01:
Not an issue. A name cannot be the first ID by construction.
The first ID is the first object built in UHDM, typically a package or module.
Starting with ID 0 is correct.

closed 2020-07-01 by alaindargelas