chipsalliance/UHDM #299: Multiple vpiXXX with the same name in a yaml file: what does that mean ?

hzeller, 2020-07-06:
I have seen a few instances in which the same vpi name is used in one file and I am not entirely sure where this makes sense.

For instance [`tchk.yaml`](https://github.com/alainmarcel/UHDM/blob/master/model/tchk.yaml)

There is an [`expr`](https://github.com/alainmarcel/UHDM/blob/master/model/tchk.yaml#L23) property with `vpiDelay` name (and type expr)
and also [`limit`](https://github.com/alainmarcel/UHDM/blob/master/model/tchk.yaml#L48) property with `vpiDelay` (and type delay)

In the generated `tchk.h`, the expr shows up, but not the `limit` property. I suspect the limit was shadowed somehow in the code generation thus never showed up ?

There are other files where the same property shows up with different cardinality: in [`for.yaml`](https://github.com/alainmarcel/UHDM/blob/master/model/for.yaml) a `vpiForInitStmt` and `vpiForIncStmt` shows up as cardinality 1 and any.

Are these just typos, or is there a good future use in it ?

alaindargelas, 2020-07-07:
They are both correct in following the standard. See models:
https://github.com/alainmarcel/UHDM/blob/master/third_party/Verilog_Object_Model.pdf

For the for loop there are 2 cases of cardinality in the model, the "s" in the name distinguishes them.
Cardinality 1 will be accessed with vpi_get
Caridinality any will be accessed with vpi_scan

For the tchk,
the expr will be accessed by vpi_get(vpiDelay, h)
the limit will be accessed by vpi_get_delays(...)

It's all good.

closed 2020-07-07 by alaindargelas

hzeller, 2020-07-07:
Ok cool, thanks for the clarification.