chipsalliance/UHDM #305: vpiSize not parsed if multiple variables declared with same data type

anadishukla, 2020-07-10:
Expected to have vpiSize defined for input variable "b" which has same property as "a".
This is in accordance to verilog and systemverilog references. 

**VERILOG Code:** 
module add(
  input [31:0] a,b,
  output signed [31:0] sum
);
  assign sum = a + b ;
endmodule

**UHDM equivalent code:** 
====== UHDM =======
design: (work@add)
|vpiName:work@add
|uhdmallPackages:
\_package: builtin, parent:work@add
  |vpiDefName:builtin
  |vpiFullName:builtin
|uhdmallClasses:
\_class_defn: (builtin::array)
  |vpiName:builtin::array
  |vpiFullName:builtin::builtin::array
|uhdmallClasses:
\_class_defn: (builtin::queue)
  |vpiName:builtin::queue
  |vpiFullName:builtin::builtin::queue
|uhdmallClasses:
\_class_defn: (builtin::string)
  |vpiName:builtin::string
  |vpiFullName:builtin::builtin::string
|uhdmallClasses:
\_class_defn: (builtin::system)
  |vpiName:builtin::system
  |vpiFullName:builtin::builtin::system
|uhdmallModules:
\_module: work@add, file:/home/anadi/Desktop/synthesis_code/surelog-inferencing/comparer/test_dir/verilog/test_signed/test.v, line:1, parent:work@add
  |vpiDefName:work@add
  |vpiFullName:work@add
  |vpiPort:
  \_port: (a), line:2
    |vpiName:a
    |vpiDirection:1
    |vpiLowConn:
    \_ref_obj: 
      |vpiActual:
      \_logic_net: (a), line:2
        |vpiName:a
        |vpiFullName:work@add.a
  |vpiPort:
  \_port: (b), line:2
    |vpiName:b
    |vpiDirection:1
    |vpiLowConn:
    \_ref_obj: 
      |vpiActual:
      \_logic_net: (b), line:2
        |vpiName:b
        |vpiFullName:work@add.b
  |vpiPort:
  \_port: (sum), line:3
    |vpiName:sum
    |vpiDirection:2
    |vpiLowConn:
    \_ref_obj: 
      |vpiActual:
      \_logic_net: (sum), line:3
        |vpiName:sum
        |vpiFullName:work@add.sum
  |vpiContAssign:
  \_cont_assign: , line:6
    |vpiRhs:
    \_operation: , line:6
      |vpiOpType:24
      |vpiOperand:
      \_ref_obj: (a), line:6
        |vpiName:a
        |vpiFullName:work@add.a
      |vpiOperand:
      \_ref_obj: (b), line:6
        |vpiName:b
        |vpiFullName:work@add.b
    |vpiLhs:
    \_ref_obj: (sum), line:6
      |vpiName:sum
      |vpiFullName:work@add.sum
  |vpiNet:
  \_logic_net: (a), line:2
  |vpiNet:
  \_logic_net: (b), line:2
  |vpiNet:
  \_logic_net: (sum), line:3
|uhdmtopModules:
\_module: work@add (work@add), file:/home/anadi/Desktop/synthesis_code/surelog-inferencing/comparer/test_dir/verilog/test_signed/test.v, line:1
  |vpiDefName:work@add
  |vpiName:work@add
  |vpiPort:
  \_port: (a), line:2, parent:work@add
    |vpiName:a
    |vpiDirection:1
    |vpiLowConn:
    \_ref_obj: 
      |vpiActual:
      \_logic_net: (a), line:2, parent:work@add
        |vpiName:a
        |vpiFullName:work@add.a
        |vpiRange:
        \_range: , line:2
          |vpiLeftRange:
          \_constant: , line:2
            |vpiConstType:7
            |vpiDecompile:31
            |vpiSize:32
            |INT:31
          |vpiRightRange:
          \_constant: , line:2
            |vpiConstType:7
            |vpiDecompile:0
            |vpiSize:32
            |INT:0
  |vpiPort:
  \_port: (b), line:2, parent:work@add
    |vpiName:b
    |vpiDirection:1
    |vpiLowConn:
    \_ref_obj: 
      |vpiActual:
      \_logic_net: (b), line:2, parent:work@add
        |vpiName:b
        |vpiFullName:work@add.b
  |vpiPort:
  \_port: (sum), line:3, parent:work@add
    |vpiName:sum
    |vpiDirection:2
    |vpiLowConn:
    \_ref_obj: 
      |vpiActual:
      \_logic_net: (sum), line:3, parent:work@add
        |vpiName:sum
        |vpiFullName:work@add.sum
  |vpiContAssign:
  \_cont_assign: , line:6
    |vpiRhs:
    \_operation: , line:6
      |vpiOpType:24
      |vpiOperand:
      \_ref_obj: (a), line:6
        |vpiName:a
        |vpiFullName:work@add.a
        |vpiActual:
        \_logic_net: (a), line:2, parent:work@add
      |vpiOperand:
      \_ref_obj: (b), line:6
        |vpiName:b
        |vpiFullName:work@add.b
        |vpiActual:
        \_logic_net: (b), line:2, parent:work@add
    |vpiLhs:
    \_ref_obj: (sum), line:6
      |vpiName:sum
      |vpiFullName:work@add.sum
      |vpiActual:
      \_logic_net: (sum), line:3, parent:work@add
  |vpiNet:
  \_logic_net: (a), line:2, parent:work@add
  |vpiNet:
  \_logic_net: (b), line:2, parent:work@add
  |vpiNet:
  \_logic_net: (sum), line:3, parent:work@add
===================

alaindargelas, 2020-07-15:
Fixed by https://github.com/alainmarcel/Surelog/pull/629.

Please open this kind of issues in Surelog, not in UHDM. The UHDM model didn't need to be changed. It was the Parser (Surelog) not populating the model correctly.

closed 2020-07-15 by alaindargelas

anadishukla, 2020-07-16:
Thanks Alain for fix. Moving forward I will file such bugs in Surelog.