chipsalliance/UHDM #306: Surelog and Verible

simdream, 2020-07-15:
Just a quick question, is Surelog and Verible (both supported with UHDM) serve the same purpose of SV frontend parser?

alaindargelas, 2020-07-16:
Verible is intended more as a SystemVerilog text formater and syntax linter.
Verible is more intended as an end-application in itself.

Surelog is a full SystemVerilog compiler that represents all the Design and Testbench in a VPI-like (UHDM) database.
Surelog/UHDM is more intended as an intermediate step for Synthesis (we have an integration with Yosys) or for Simulation (We have an integration with Verilator) for now (https://github.com/alainmarcel/uhdm-integration).
There are a few users who also use Surelog/UHDM for other applications not yet in public domain. 

What are you looking for exactly? That might help us give you a more educated choice for a front-end.

simdream, 2020-07-17:
We are evaluating Verialtor, with the purpose of speed up the simulating further by partitioning the design into several verilated IPs. IPs can run in parallel. With that, we have also be looking at other SV parser/elaboration alternatives. The most recent DSL such as Chisel could possibly be potential but for sure is not the main design language. So Surelog's able to start from SV definitely is quite attractive. 

One thing I am not sure is, what is the  benefit of UHDM, as this is similar to FIRRTL anyway. 

Another question is, if we use Verilator, what is the point to cur off its front end flow and import design from UHDM format. Since UHDM is quite new for me, I did not find good material tells what exactly its format is, or we can say, similar to FIRRTL, or AST (the tree or XML output from Verilator).

alaindargelas, 2020-07-17:
UHDM is similar to FIRRTL in a loose manner:
UHDM is actually trying to stick as close to the SystemVerilog VPI API Standard as much as possible.
I did not produce a documentation because the Standard is the documentation.
You can find the following links at the bottom of the UHDM main page:
Verilog_Object_Model.pdf - Object Model section of the IEEE_Std1800-2017
SystemVerilog 2017 - System Verilog Standard

All the objects in the Design and Testbench (UVM/Constraints....) will be accessible through UHDM.

We already have a UHDM->Verilator repo for the purpose of augmenting Verilator support of all the SystemVerilog features.
If your purpose is to partition the Design for Verilator, you could contribute to that existing repo (or fork from it) your partitionner code (That would be using UHDM API).

simdream, 2020-07-17:
Will fork one and start with one example to understand. For verilator itself support UVM might be a long way to go since the delays are not quite supported yet. But to keep UVM and the testbench outside of Verilator could be an option, a co-simulation environment.

closed 2020-07-17 by simdream