chipsalliance/UHDM #319: Support a second parser (other than Surelog)

mithro, 2020-07-24:
The goal of UHDM is to allow "users" (like Yosys, Verilator and Icarus) of a SystemVerilog parsers not to care about the exact parser implementation. However, currently the only parser which exposes the UHDM interface is Surelog.

Adding a second parser which also exposed the UHDM API would help verify that the API isn't over fitted to Surelog in any way.

mithro, 2020-07-24:
I have a strong preference for that second parser to be [Verible](https://github.com/google/verible) but I don't believe we have the development resources to do that work at the moment?

mithro, 2020-07-24:
FYI - @hzeller

MikePopoloski, 2021-08-16:
I started looking at this and had some questions about the model. Normally when using VPI the model is entirely the instantiated design, so each "module" is actually an instantiated instance with all parameter values set. When looking at, say, the genscope_cells example from the uhdm-integration repo, what I see is a bit confusing: there is an allModules array which has only one copy of the ibex_counter module even though there are two instances in the elaborated design. Is Surelog just picking one of the instances at random (probably the first occurrence) and throwing it into the "allModules" array? How does the tool decide which parts of an instance go in the "allModules" module and which go in the "vpiModule" entry within the genscope array? Are consumers expected to go back and re-elaborate the code in the allModules definition for each instance based on the actual param values?

Unrelated, but also wondering why all of the variables seem to be serialized as nets, with a vpiNetType of 36 which isn't one of the listed nettypes in the spec...

hzeller, 2021-08-16:
@alaindargelas ^ some questions for you

alaindargelas, 2021-08-16:
UHDM has a flat list of modules, interfaces, packages classes (Named allModules....) and the Elaborated list of topmodules called topModules with all their respective hierarchy elaborated.
The allModules is not used by any client applications yet as far as a know, although the concept of having an unelaborated view (modeling the unaltered lexical description of the modules) has it's use in linting, it is not the main goal and you can ignore it.

The topModules does contain all the elaboration users need and they do not need to perform extra elaboration task.
To be noted, in Surelog, the option -elabuhdm does not change the hierarchy structure (Even without this option, all generate statements are unrolled, but this option, or using the programmatic call to uhdm elaboration does uniquify the nets, functions and other objects that can be impacted by parameters.
To date the UHDM elaborated model does not reconcile lhs and rhs expression padding (And eventually that might get addressed), as both Verilator and Yosys backends do perform this operation.
Also the drivers/loads are not computed yet (A typical VPI implementation in a simulator does).

"Unrelated, but also wondering why all of the variables seem to be serialized as nets, with a vpiNetType of 36 which isn't one of the listed nettypes in the spec..."
The allModules view pays little attention to net types, the elaborated view sets all the right net types.

MikePopoloski, 2021-08-16:
The instances in topModules does not seem to contain all of the information in the module though. For example, I don't seen any vpiProcesses and associated statements in there, or actually anything from the inside the module itself except for nested generate scopes. I'm using uhdm-dump on the output of the genscope_cells test by the way to inspect the database.

Also as far as I can tell the variables are serialized as nets in the topModules tree as well, it doesn't look any different to me.

alaindargelas, 2021-08-17:
I forgot, yes all the processes are not elaborated, so they reside in non-elaborated form in the allModules. Yosys and Verilator (The main targets for UHDM for now) do perform the final uniquification of the processes. The respective UHDM-Yosys UHDM-Verilator plugins do pass the elaborated tree from UHDM and the non-elaborated parts from the flat modules.

If you check examples like:
https://github.com/chipsalliance/Surelog/blob/master/tests/OneNet/OneNet.log
You should see some different types in the elaborated view.
Please open an Issue in Surelog with an example.

MikePopoloski, 2021-08-17:
Ah, ok. Is there documentation of what gets elaborated and what does not? It's pretty confusing as-is. I see that continuous assigns don't show up in topModules, unless they're in a generate scope and then they do? Or is the model still a work in progress and not ready for another implementation?

alaindargelas, 2021-08-17:
I'll write up a complete description of the elaboration in the README
shortly.
Essentially, we are trying to keep the model as folded as possible, until
it really needs to be uniquified. The examples you give are an illustration
of that.
The allModules will contain the processes and assigns at the top module and
placeholder ports and nets.
The topModules will contain the hierarchy, the elaborated nets with the
correct type. the generate statements, the logic generated in the generate
statements...

The plugins for Verilator and Yosys do a bit of assembly. Making the model
fully elaborated, is actually quite trivial (clone everything all the
time), but the memory required is quite large and make the downstream
plugins have to explicitly translate every object to the native data
structures of Yosys and Verilator which in turn do some elaboration (memory
bloat).
Keeping the model folded saves on the plugin memory footprint, also des not
replicate what the client applications are doing in their internal
elaboration anyway.

But we could easily create an optional fully elaborated model (with all the
objects cloned from allModules to the topModules for regression purposes,
say to compare the UHDM generated model in between our 2 implementations.

MikePopoloski, 2021-08-17:
Thanks, that makes sense. Have you thought about alternative models that would more compactly represent the elaborated design? It seems like if you require all downstream consumers to reimplement all of the type checking and propagation logic it somewhat defeats the purpose of UHDM being the full frontend, as that logic is quite complicated. You say that it's simple but I don't see how it could be; changing the parameter values for a module can result in wildly different types involved in each expression of a process / continuous assign.

alaindargelas, 2021-08-17:
UHDM is a R/W model, so all the uniquification/type checking can be done
and annotated/substituted in the model itself by subsequent passes.
For Instance the -elabhm option of Surelog simply calls the UHDM Elaborator
class that clone UHDM objects (Say an assign stmt in a generate block and
bind the variables to the variables in the scope of the generate). That is
done by reading UHDM and writing back into UHDM. All the constant
propagation required to perform elaboration is done by Surelog, but
expressions that are "runtime" expressions are left unevaluated. One can
write an Evaluator on top of UHDM and decide to perform more evaluation
based on the UHDM model.
Similarly with the load/driver data, that can be computed from UHDM and
annotated back into UHDM.
All of these middle layer applications are in the TODO list.

To date I see 2 types of customer for UHDM,
- the Yosys/Verilator applications that already perform the additional
tasks above (So not an emergency for them),
- The lighter weight applications like State machine analyzer, netlist
viewer, linters that would benefit from these extra services but who
already make good use of the folded model.

alaindargelas, 2021-09-04:
@MikePopoloski, coming back to this after a while, sorry...
I changed the README to have more explanations.
I also realized the the Antmicro team is not using the -elabuhdm Surelog option which does bloat the memory, but does what you are looking for.
That option (modulo bugs I'd be happy to fix) does indeed clone all objects including processes.
So on the example you are pointing out,
https://github.com/chipsalliance/UHDM-integration-tests/blob/master/tests/genscope_cells/dut.sv

surelog -parse -elabuhdm -d uhdm  -d vpi_ids

Produces the following fully elaborated tree (including processes and assigns):
```
....
|uhdmtopModules:
\_module: work@dut (work@dut), id:444 tests/ScratchPad.sv:30:1: , endln:55:10
  |vpiDefName:work@dut
  |vpiName:work@dut
  |vpiGenScopeArray:
  \_gen_scope_array: (work@dut.gen_modules[0]), id:465, line:41:41, endln:51:4, parent:work@dut, parID:444
    |vpiName:gen_modules[0]
    |vpiFullName:work@dut.gen_modules[0]
    |vpiGenScope:
    \_gen_scope: (work@dut.gen_modules[0]), id:466, parent:work@dut.gen_modules[0], parID:465
      |vpiFullName:work@dut.gen_modules[0]
      |vpiModule:
      \_module: work@ibex_counter (work@dut.gen_modules[0].module_in_genscope), id:467 tests/ScratchPad.sv:42:3: , endln:50:35, parent:work@dut.gen_modules[0], parID:466
        |vpiDefName:work@ibex_counter
        |vpiDefFile:tests/ScratchPad.sv
        |vpiDefLineNo:57
        |vpiName:module_in_genscope
        |vpiFullName:work@dut.gen_modules[0].module_in_genscope
        |vpiPort:
        \_port: (clk_i), id:676, line:60:23, endln:60:28, parent:work@dut.gen_modules[0].module_in_genscope, parID:467
          |vpiName:clk_i
          |vpiDirection:1
          |vpiHighConn:
          \_ref_obj: (work@dut.gen_modules[0].clk), id:677, line:43:42, endln:43:45, parent:clk_i, parID:676
            |vpiName:clk
            |vpiFullName:work@dut.gen_modules[0].clk
            |vpiActual:
            \_logic_net: (work@dut.clk), id:128, line:34:7, endln:34:10, parent:work@dut, parID:444
              |vpiName:clk
              |vpiFullName:work@dut.clk
              |vpiNetType:36
          |vpiLowConn:
          \_ref_obj: (work@dut.gen_modules[0].module_in_genscope.clk_i), id:678, parent:clk_i, parID:676
            |vpiName:clk_i
            |vpiFullName:work@dut.gen_modules[0].module_in_genscope.clk_i
            |vpiActual:
            \_logic_net: (work@dut.gen_modules[0].module_in_genscope.clk_i), id:513, line:60:23, endln:60:28, parent:work@dut.gen_modules[0].module_in_genscope, parID:467
              |vpiName:clk_i
              |vpiFullName:work@dut.gen_modules[0].module_in_genscope.clk_i
              |vpiNetType:36
          |vpiInstance:
          \_module: work@ibex_counter (work@dut.gen_modules[0].module_in_genscope), id:467 tests/ScratchPad.sv:42:3: , endln:50:35, parent:work@dut.gen_modules[0], parID:466
        |vpiPort:
        \_port: (rst_ni), id:679, line:61:23, endln:61:29, parent:work@dut.gen_modules[0].module_in_genscope, parID:467
          |vpiName:rst_ni
          |vpiDirection:1
          |vpiHighConn:
          \_ref_obj: (work@dut.gen_modules[0].rst), id:680, line:44:43, endln:44:46, parent:rst_ni, parID:679
            |vpiName:rst
            |vpiFullName:work@dut.gen_modules[0].rst
            |vpiActual:
            \_logic_net: (work@dut.rst), id:129, line:35:7, endln:35:10, parent:work@dut, parID:444
              |vpiName:rst
              |vpiFullName:work@dut.rst
              |vpiNetType:36
          |vpiLowConn:
          \_ref_obj: (work@dut.gen_modules[0].module_in_genscope.rst_ni), id:681, parent:rst_ni, parID:679
            |vpiName:rst_ni
[... 1800 lines cut ...]
    |vpiRhs:
    \_constant: , id:125, line:32:29, endln:32:30
      |vpiConstType:9
      |vpiDecompile:2
      |vpiSize:64
      |UINT:2
      |vpiTypespec:
      \_int_typespec: (NUM_MODULES), id:0, line:32:11, endln:32:14, parent:work@dut.NUM_MODULES, parID:1
        |vpiName:NUM_MODULES
    |vpiLhs:
    \_parameter: (work@dut.NUM_MODULES), id:445, line:32:15, endln:32:30, parent:work@dut, parID:444
  |vpiParamAssign:
  \_param_assign: , id:126, line:33:15, endln:33:32, parent:work@dut, parID:444
    |vpiRhs:
    \_constant: , id:127, line:33:30, endln:33:32
      |vpiConstType:9
      |vpiDecompile:10
      |vpiSize:64
      |UINT:10
      |vpiTypespec:
      \_int_typespec: (MODULE_PARAM), id:4, line:33:11, endln:33:14, parent:work@dut.MODULE_PARAM, parID:5
        |vpiName:MODULE_PARAM
    |vpiLhs:
    \_parameter: (work@dut.MODULE_PARAM), id:447, line:33:15, endln:33:32, parent:work@dut, parID:444
```

MikePopoloski, 2021-09-05:
No problem, thanks for following up. I think I understand the two different modes now, and the additions to the README are helpful. I'll ask more questions as I run into them.

Also as a side note, my original point about the nets instead of variables still stands. The original code declares a variable (just picking one as an example):
```
logic [CounterWidth-1:0] counter_d;
```
but the UHDM dump has it as a net instead:
```
vpiNet:
        \_logic_net: (work@dut.gen_modules[0].module_in_genscope.counter_d), id:522, line:74:28, endln:74:37, parent:work@dut.gen_modules[0].module_in_genscope, parID:467
```

alaindargelas, 2021-09-05:
Got it now. Since Surelog does not do driver/load analysis as of now, it can't know if he usage of a variable and assumes everything that is declared as a logic is a net. 
I'm not sure the standard is very explicit here, but for something to be a variable and not a net it has to be a local usage (hence you have to perform some form of traversal to figure it out). Lacking that knowledge, Surelog assumes all logic types are nets.
If you have a better semantic interpretation that does not require a traversal of the driver/load please share! I'll be happy to implement the same

MikePopoloski, 2021-09-05:
Oh, it's much simpler than that. A net declaration has a net type specified, and otherwise it's a variable. logic is a data type, not a net type, so the declaration is a variable. Net types are things like "wire", "tri", etc. You do need to handle user-defined nettypes but they are very rare in practice.

alaindargelas, 2021-09-05:
OK, let's spec it out:
wire a;            // net
wire logic b; // net
logic c;           // var logic
var d;             // var logic

Page 704 of the 2017 standard has a lots of example where ports of type wire are equivalent to "wire logic".
module mh0 (wire x); // inout wire logic x
module mh1 (integer x); // inout wire integer x
module mh2 (inout integer x); // inout wire integer x
module mh3 ([5:0] x); // inout wire logic [5:0] x
module mh4 (var x); // ERROR: direction defaults to inout which cannot be var
module mh5 (input x); // input wire logic x
module mh6 (input var x); // input var logic x
module mh7 (input var integer x); // input var integer x
module mh8 (output x); // output wire logic x
module mh9 (output var x); // output var logic x
module mh10(output signed [5:0] x); // output wire logic signed [5:0] x
module mh11(output integer x); // output var integer x
module mh12(ref [5:0] x); // ref var logic [5:0] x
module mh13(ref x [5:0]); // ref var logic x [5:0]
module mh14(wire x, y[7:0]); // inout wire logic x, inout wire logic y[7:0]
module mh15(integer x, signed [5:0] y); // inout wire integer x,  inout wire logic signed [5:0] y
module mh16([5:0] x, wire y); // inout wire logic [5:0] x, inout wire logic y
module mh17(input var integer x, wire y); // input var integer x, input wire logic y
module mh18(output var x, input y); // output var logic x, input wire logic y
module mh19(output signed [5:0] x, integer y);  // output wire logic signed [5:0] x, output var integer y
module mh20(ref [5:0] x, y); // ref var logic [5:0] x, ref var logic [5:0] y
module mh21(ref x [5:0], y); // ref var logic x [5:0], ref var logic y

MikePopoloski, 2021-09-05:
Yes, that's right. The rules for ports are more complicated as you mention but still fully determined by the declaration itself and not how they are used. 23.2.2.3 has the algorithm for determining net vs variable for ports when not explicitly specified.

alaindargelas, 2021-09-07:
The following put request: "fix net var to match standard chipsalliance/Surelog#1809" should fix this issue.
I created a test that tests all the major possibilities:
https://github.com/chipsalliance/Surelog/tree/master/tests/StandardNetVar

mysoreanoop, 2021-11-04:
> @MikePopoloski, coming back to this after a while, sorry...
> I changed the README to have more explanations.
> I also realized the the Antmicro team is not using the -elabuhdm Surelog option which does bloat the memory, but also does what you are looking for.
> That option (modulo bugs I'd be happy to fix) does indeed clone all objects including processes.
> So on the example you are pointing out,
....

I'm trying to replicate the kind of output you got here -- with the scope hierarchy. Using `uhdm-dump` surelog.uhdm, I'm able to see it. I am doing pretty much the same thing done in UHDM/util/uhdm-dump.cpp in Surelog/src/hellouhdm.cpp -- that is vpi_iterate through vpiNets and use `vpi_get_str(vpiFullName, <handle of ref_obj>); ` All I get are <module name>.<local names of variables>

Any idea what could be going wrong?

alaindargelas, 2021-11-05:
You are navigating the vpiAllModules instead of the vpiTopModules top level iterator.
From the vpiTopModules, you can recusvely traverse the sub instance and get handles on signals. You can use the vpiFullName attribute or you can also build hierarchical names yourself while doing the traversal (Keep track of your hierarchy instance name).

alaindargelas, 2021-11-05:
If you send snippet of code, it would be easier to point to the issues.

mysoreanoop, 2021-11-06:
Yeah, works with uhdmtopModules. So, just to conclude:
We gotta walk the uhdmtopModules tree for full hierarchical names of nets/regs. If we also need, say, all the ternary statements these variables appear in, we'd have to switch to the uhdmallModules and look for continuous and procedural assignments that use ternaries, and get the local names of nets/regs, but save the corresponding full names from uhdmtopModules. Does that sound correct? 

If so, is there any {helper code/advice} you could {point to/give us}?

Thank you so much!

alaindargelas, 2021-11-06:
You got it.
If you want to avoid going back and forth, you can invoke the optional UHDM uniquification phase:
  - In Surelog option -uhdmelab, or 
  - after reloading UHDM, invoke the UHDM elaborator in your main: (2 lines of code) https://github.com/alainmarcel/UHDM/blob/4926f83bfe5b8d7f16ffd9572e280e6236fca25c/util/uhdm-dump.cpp#L131)

That way you can navigate the uhdmTopModules and don't need to go to the uhdmAllModules. Memory is larger or course, but all types and sizes are uniquified.

mysoreanoop, 2021-11-06:
Wow, neat! Thank you