chipsalliance/UHDM #374: linux compilation fails from a path that contains link

DamienLeBars, 2020-12-01:
the use of [file normalize XXX] (resolving links) in model_gen.tcl generate an error in CAPNP call that use path without links resolving.

"File is not in the current directory and does not match any prefix defined with "
 "--src-prefix. Please pass an appropriate --src-prefix so I can figure out where to "
 "write the output for this file."

alaindargelas, 2020-12-02:
I'm not seeing this problem, neither the Actions or Travis builds.
Are you sure there is not something else at play here?

DamienLeBars, 2020-12-02:
yes, on our network, we have user home path that contains link.
daisy is the computer i'm connected on and /scorpion is a link to user shared home disk
```bash
daisy:/scorpion/home/dlb > cd
daisy:/scorpion/home/dlb > pwd
/net/scorpion-home/Scorpion-home/home/dlb
daisy:/scorpion/home/dlb > echo $PWD
/scorpion/home/dlb
```
So capnp see the path like $PWD when [file normalize] return resolved path like pwd.
I understand that it is a quite specific problem!

alaindargelas, 2020-12-02:
I see.

DamienLeBars, 2020-12-03:
Fixed by [alainmarcel/UHDM#375](https://github.com/alainmarcel/UHDM/pull/375)

closed 2020-12-03 by DamienLeBars