chipsalliance/UHDM #426: Can I synthesis my systemverilog design through Yosys by taking help from UHDM?

zeeshanrafique23, 2021-06-02:
I have made few amendments in ibex core and the design is still *.sv and I am having an issue with synthesizing the design on Yosys, I want to know if this tool could do this for me or not?

hzeller, 2021-06-02:
There is work-in-progress to use https://github.com/alainmarcel/uhdm-integration to read UHDM in Yosys and process system verilog.
Ibex was possible to process, but since it is work in progress, @kbieganski can probably help with details to get things going.

tgorochowik, 2021-06-03:
Like Henner said - the goal in this case is indeed to be able to process and then synthesize everything, but at this point we cannot promise that what we already have will be enough for you.

Could you share some more details about the problems you're having? We may be able to help if it's some general issue with tooling or something, if it's a matter of lack of support for some language constructs (if sv2v didn't help I guess they could be) then it may be hard to resolve this easily, but we'd certainly like to at least gather feedback about what users are using and wish we supported

alaindargelas, 2021-06-03:
@zeeshanrafique23, the immediate thing you can do is submit a snippet of SystemVerilog code that does not work in sv2v, we can analyze it and make sure Surelog/UHDM process it correctly.

zeeshanrafique23, 2021-06-03:
The issue is that there is an input parameter of type 'Type' [here](https://github.com/merledu/azadi_apr/blob/8f9580a7192bc1922c956fc2b7681eb822c2f876/azadi_rtl_v0.6/rr_arb_tree.sv#L53), through which we can pass the structure as type.
I have done the hard wire of this parameter as this module "[rr_arb_tree](https://github.com/merledu/azadi_apr/blob/main/azadi_rtl_v0.6/rr_arb_tree.sv)" is called only two times in two different modules.  In [fpnew_top.sv](https://github.com/merledu/azadi_apr/blob/8f9580a7192bc1922c956fc2b7681eb822c2f876/azadi_rtl_v0.6/fpnew_top.sv#L147) and [fpnew_opgroup_block.sv](https://github.com/merledu/azadi_apr/blob/8f9580a7192bc1922c956fc2b7681eb822c2f876/azadi_rtl_v0.6/fpnew_opgroup_block.sv#L204).

How I hard wire: as the module is called in two module so I made two modules named rr_arb_tree_opgroup_block and rr_arb_tree_fpnew_top in file [azadi_soc_csv.v](https://github.com/zeeshanrafique23/azadi-netlist/blob/master/azadi_soc_csv.v) line number 29311 and 29607, and they are instantiated on line number 24907 and 25858 respectively, and added the structure type in 'packege fpnew_pkg' on line number 1345 to directly call the package where the key word "Type was used in 'rr_arb_tree'".
Still the issue arises after these changes is in converted verilog file [azadi_soc_top.v](https://github.com/zeeshanrafique23/azadi-netlist/blob/master/azadi_soc_top.v), the input and output (data_i and data_o -like in line 26058 of azadi_soc_top.v) in the modules rr_arb_tree* are genereted are structure type and icarus verilog in giving error message on it.

zeeshanrafique23, 2021-06-03:
Further I am going to try it by expanding the structure on input output ports, only this 'rr_arb_tree' is the module which is causing the problem until now.

alaindargelas, 2021-06-04:
Starting looking at it

alaindargelas, 2021-06-04:
Surelog parses and elaborates correctly now, but does not override the type param correctly. I'll fix this tomorrow.

alaindargelas, 2021-06-05:
Surelog overrides the parameter correctly. This case needs to move now to https://github.com/alainmarcel/uhdm-integration

alaindargelas, 2021-06-05:
@tgorochowik, the source is setup in Surelog here: https://github.com/alainmarcel/Surelog/tree/master/third_party/tests/AzadiRTL.
The steps would be to:
- Try the Yosys compilation does not error out,
- Instruct @zeeshanrafique23 on how to operate uhdm-integration and try on hardware, as you probably don't have a mean to verify the result is working in hardware (There is no test with the source tree).

closed 2021-06-05 by alaindargelas