chipsalliance/UHDM #427: segmentation fault in UHDM frontend to synthesize OpenTitan

sarahSherifAli, 2021-06-09:
getting segmentation fault while executing UHDM frontend to synthesize OpenTitan 
attached below the makefile, makefile.in yosys script yosys executable and UHDM library, the source of the opentitan files and a screen shot form gdb shows the last frame that throws the seg fault 
[seg_fault_uhdm_titan.tar.gz](https://github.com/alainmarcel/UHDM/files/6622306/seg_fault_uhdm_titan.tar.gz)
[image]

alaindargelas, 2021-06-10:
@sarahSherifAli I think there are several issues in your Makefiles which I'm not going to debug.
Please try this baseline project first which is a Surelog command line adapted from your Makefile.in
In your translation of the Earlgrey.sl file from Surelog you must have introduced errors. One for instance is the name of the top level module that does not exist in the design.
Please do the following in 
cd Earlgrey_nexysvideo
tar xvzf sarah.tar.gz 
surelog -f sarah.prj
uhdm-dump slpp_all/surelog.uhdm

That should work. After that it's a matter of making sure your Makefiles do achieve the same results with Surelog and uhdm-dump,
Finally, try again in the Yosys integration and report back any more issues.

[sarah.tar.gz](https://github.com/alainmarcel/UHDM/files/6628383/sarah.tar.gz)

alaindargelas, 2021-06-12:
@sarahSherifAli, can we close this issue? Are there any more problems?

closed 2021-06-15 by alaindargelas