chipsalliance/UHDM #480: Garbage collect UHDM when saving

alaindargelas, 2021-09-02:
A lot of temporary UHDM object are created while performing the Elaboration in Surelog.
These objects should not be serialized on disk, only the objects that are accessible from the designs handles should be serialized.

hzeller, 2021-09-02:
Can you give me entry points (links to code on github) where the save is happening and an examples of where a temporary object is created, or a temporary object is abandoned (I could imagine 'abandoning' might happen if we have some node like `a + b` which then is replaced by the sum).

Do you have an example of a particularly _large_ test case that allocates lots of memory that I can run on my machine to play around with ideas ? I suspect you have some of these in the Surelog test-suite, so if you can give me a complete command line how to run one of these, would be very helpful.

My initial idea would be to intrinsic reference counting (i.e. the counter is inside the UHDM object and it `delete this;` when it goes to zero), which would make that simple. It would require that we don't have circles back, but I'd suspect everything is represented as DAG.

Anyway, need to stare at some examples to form some opinion towards a solution.

alaindargelas, 2021-09-03:
I created a new unit test that demonstrate this with the minimum set of objects,
a design, a module part of the design and a module not part of the design.
The latter is the one that can be seen as "temporary", it is serialized but it should not. 

```
~/UHDM$ ./build/bin/test_garbage_collect 
Make design
VISITOR:
design: (design1)
|vpiName:design1
|uhdmtopModules:
\_module: M1 (M1), parent:design1
  |vpiTopModule:1
  |vpiDefName:M1
  |vpiFullName:M1

Save design
Restore design
decompile:
design: (design1)
|vpiName:design1
|uhdmtopModules:
\_module: M1 (M1), parent:design1
  |vpiTopModule:1
  |vpiDefName:M1
  |vpiFullName:M1

decompile:
module: M1 (M1), parent:design1
|vpiTopModule:1
|vpiDefName:M1
|vpiFullName:M1

decompile:
module: M2 (u1)
|vpiDefName:M2
|vpiName:u1

VISITOR:
design: (design1)
|vpiName:design1
|uhdmtopModules:
\_module: M1 (M1), parent:design1
  |vpiTopModule:1
  |vpiDefName:M1
  |vpiFullName:M1

ALL OBJECTS:
OBJECT:
decompile:
design: (design1)
|vpiName:design1
|uhdmtopModules:
\_module: M1 (M1), parent:design1
  |vpiTopModule:1
  |vpiDefName:M1
  |vpiFullName:M1
OBJECT:
decompile:
module: M1 (M1), parent:design1
|vpiTopModule:1
|vpiDefName:M1
|vpiFullName:M1
OBJECT:
decompile:
module: M2 (u1)            <---------------- This object is not part of the design, it should not have been serialized in the first place
|vpiDefName:M2
|vpiName:u1
```

hzeller, 2021-09-11:
The `test_garbage_collect` test has something going on: it crashes on MacOS, and in fact running it on Linux with valgrind also that there is a read on something already deleted. Will investigate while looking at this, but for now I am going to disable that test, as it breaks the CI possibly hiding something else.

hzeller, 2022-06-15:
With a leak address sanitizer I see these memory leaks when running the `garbage_collect_test.cpp`

```
Direct leak of 24 byte(s) in 1 object(s) allocated from:
    #0 0x564147fc06dd in operator new(unsigned long) llvm/lib/asan/asan_new_delete.cpp:95:3
    #1 0x7effb595bb60 in Make genfiles/UHDM/uhdm/BaseClass.h:160:18
    #2 0x7effb595bb60 in UHDM::Serializer::RestoreAdapter::operator()(Design::Reader, UHDM::Serializer*, UHDM::design*) const genfiles/UHDM/src/Serializer_restore.cpp:4051:64
    #3 0x7effb56e5c3e in operator()<UHDM::design, Design, void> genfiles/UHDM/src/Serializer_restore.cpp:4065:7
    #4 0x7effb56e5c3e in UHDM::Serializer::Restore(std::__u::basic_string<char, std::__u::char_traits<char>, std::__u::allocator<char>> const&) genfiles/UHDM/src/Serializer_restore.cpp:4483:20
    #5 0x7effb62e0c92 in GarbageCollectTest_NoLeakExpectation_Test::TestBody() UHDM/tests/garbage_collect_test.cpp:51:62
    #6 0x7eff681a17c8 in void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) gtest/googletest/src/gtest.cc
    #7 0x7eff68174d94 in testing::Test::Run() gtest/googletest/src/gtest.cc:2674:5
    #8 0x7eff68176424 in testing::TestInfo::Run() gtest/googletest/src/gtest.cc:2853:11
    #9 0x7eff681777ab in testing::TestSuite::Run() gtest/googletest/src/gtest.cc:3012:30
    #10 0x7eff68195fbf in testing::internal::UnitTestImpl::RunAllTests() gtest/googletest/src/gtest.cc:5873:44
    #11 0x7eff681a27a8 in bool testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) gtest/googletest/src/gtest.cc
    #12 0x7eff68195547 in testing::UnitTest::Run() gtest/googletest/src/gtest.cc:5447:10
    #13 0x564147fc319d in RUN_ALL_TESTS gtest/googletest/include/gtest/gtest.h:2308:73
    #14 0x564147fc319d in main testing/base/internal/gunit_main.cc:86:10
    #15 0x7eff2ba4c8d2 in __libc_start_main (/usr/grte/v5/lib64/libc.so.6+0x628d2) (BuildId: 7cfed7708e5ab7fcb286b373de21ee76)
    #16 0x564147f06b49 in _start /usr/grte/v5/debug-src/src/csu/../sysdeps/x86_64/start.S:120

Direct leak of 24 byte(s) in 1 object(s) allocated from:
    #0 0x564147fc06dd in operator new(unsigned long) llvm/lib/asan/asan_new_delete.cpp:95:3
    #1 0x7effb56a88be in Make genfiles/UHDM/uhdm/BaseClass.h:160:18
    #2 0x7effb56a88be in Make<UHDM::module> genfiles/UHDM/src/Serializer_restore.cpp:64:19
    #3 0x7effb56a88be in UHDM::Serializer::MakeModuleVec() genfiles/UHDM/src/Serializer_restore.cpp:394:60
    #4 0x7effb62e09b6 in build_designs UHDM/tests/garbage_collect_test.cpp:30:27
    #5 0x7effb62e09b6 in GarbageCollectTest_NoLeakExpectation_Test::TestBody() UHDM/tests/garbage_collect_test.cpp:42:43
    #6 0x7eff681a17c8 in void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) gtest/googletest/src/gtest.cc
    #7 0x7eff68174d94 in testing::Test::Run() gtest/googletest/src/gtest.cc:2674:5
    #8 0x7eff68176424 in testing::TestInfo::Run() gtest/googletest/src/gtest.cc:2853:11
    #9 0x7eff681777ab in testing::TestSuite::Run() gtest/googletest/src/gtest.cc:3012:30
    #10 0x7eff68195fbf in testing::internal::UnitTestImpl::RunAllTests() gtest/googletest/src/gtest.cc:5873:44
    #11 0x7eff681a27a8 in bool testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) gtest/googletest/src/gtest.cc
    #12 0x7eff68195547 in testing::UnitTest::Run() gtest/googletest/src/gtest.cc:5447:10
    #13 0x564147fc319d in RUN_ALL_TESTS gtest/googletest/include/gtest/gtest.h:2308:73
    #14 0x564147fc319d in main testing/base/internal/gunit_main.cc:86:10
    #15 0x7eff2ba4c8d2 in __libc_start_main (/usr/grte/v5/lib64/libc.so.6+0x628d2) (BuildId: 7cfed7708e5ab7fcb286b373de21ee76)
    #16 0x564147f06b49 in _start /usr/grte/v5/debug-src/src/csu/../sysdeps/x86_64/start.S:120

Indirect leak of 8 byte(s) in 1 object(s) allocated from:
    #0 0x564147fc06dd in operator new(unsigned long) llvm/lib/asan/asan_new_delete.cpp:95:3
    #1 0x7effb58d6319 in __libcpp_operator_new<unsigned long> compiler-toolchain/include/c++/v1/new:246:10
    #2 0x7effb58d6319 in __libcpp_allocate compiler-toolchain/include/c++/v1/new:272:10
    #3 0x7effb58d6319 in allocate compiler-toolchain/include/c++/v1/__memory/allocator.h:106:38
    #4 0x7effb58d6319 in __allocate_at_least<std::__u::allocator<UHDM::module *> > compiler-toolchain/include/c++/v1/__memory/allocate_at_least.h:54:19
    #5 0x7effb58d6319 in __split_buffer compiler-toolchain/include/c++/v1/__split_buffer:325:29
    #6 0x7effb58d6319 in std::__u::vector<UHDM::module*, std::__u::allocator<UHDM::module*>>::reserve(unsigned long) compiler-toolchain/include/c++/v1/vector:1465:53
    #7 0x7effb595bbc8 in UHDM::Serializer::RestoreAdapter::operator()(Design::Reader, UHDM::Serializer*, UHDM::design*) const genfiles/UHDM/src/Serializer_restore.cpp:4052:13
    #8 0x7effb56e5c3e in operator()<UHDM::design, Design, void> genfiles/UHDM/src/Serializer_restore.cpp:4065:7
    #9 0x7effb56e5c3e in UHDM::Serializer::Restore(std::__u::basic_string<char, std::__u::char_traits<char>, std::__u::allocator<char>> const&) genfiles/UHDM/src/Serializer_restore.cpp:4483:20
[... cut ...]
    #16 0x7eff68195547 in testing::UnitTest::Run() gtest/googletest/src/gtest.cc:5447:10
    #17 0x564147fc319d in RUN_ALL_TESTS gtest/googletest/include/gtest/gtest.h:2308:73
    #18 0x564147fc319d in main testing/base/internal/gunit_main.cc:86:10
    #19 0x7eff2ba4c8d2 in __libc_start_main (/usr/grte/v5/lib64/libc.so.6+0x628d2) (BuildId: 7cfed7708e5ab7fcb286b373de21ee76)
    #20 0x564147f06b49 in _start /usr/grte/v5/debug-src/src/csu/../sysdeps/x86_64/start.S:120
```

It does look like unaccounted-for modules, though this might be a separate leaking thing.