chipsalliance/UHDM #483: Consider building against a fixed Capn Proto target

MikePopoloski, 2021-09-07:
It would be nice if UHDM targeted a specific Capn Proto release instead of living at head via a submodule, for interoperability purposes. Each commit to the upstream repo can potentially break any consumer that isn't compiled against exactly the same commit, which increases in likelihood if there are other users of UHDM besides Surelog.

It seems easier to just rely on whatever version is installed on the system (using FindPackage in cmake). On Windows users will need to build the lib themselves but that's not really any different than how it works today.

hzeller, 2021-09-08:
+1, this looks like someone with CMake-fu should be able to implement that easily.

Looking at debian packages, it looks like it comes with pkg-config packages
https://packages.debian.org/sid/amd64/libcapnp-dev/filelist

For windows users: Maybe there is also a similar devel package on chocolatey ?

hzeller, 2021-09-08:
Also, this involves to generate the capnproto outputs in the CMakefile instead of the tcl-script.

PR #856 (2023-01-14): Allow to use native capnproto.
By default, keeps using third_party/ capnproto, but allows to use the capnroto installed on the system by setting UHDM_USE_HOST_CAPNP=On
This helps UHDM to be packed in distributions.

(So wht do we keep third_party/ ? On Windows, there is no common package manager, so projects there rely more on things to be pulled in together; see discussion in #491).

Fixes: #483
commit 07cd6c68 (2023-01-14): Allow to use native capnproto.

By default, keeps using third_party/ capnproto, but allows to
use the capnroto installed on the system by setting
UHDM_USE_HOST_CAPNP=On
This helps UHDM to be packed in distributions.

(So wht do we keep third_party/ ? On Windows, there is no common
package manager, so projects there rely more on things to be pulled
in together; see discussion in #491).

Fixes: #483
closed 2023-01-14 by hzeller