chipsalliance/UHDM #484: Cleanup of header installation

hzeller, 2021-09-08:
Currently, we have two include directories that are installed:

  * `${INSTALL_PREFIX}/include/uhdm/headers`  _the generated headers_
  * `${INSTALL_PREFIX}/include/uhdm/include`   _vpi_stuff_
  
... and then there is also `uhdm.h` installed twice, in `uhdm/uhdm.h` and `uhdm/headers/uhdm.h` (#197)

The names of 'include/' and 'headers/' is very generic, so it does not convey any meaningful subdivision of functionality - they just happen to be that way because one set of files is generated.
From the users' perspective it is an arbitrary choice.

So I propose we should just have _one_ `${INSTALL_PREFIX}/include/uhdm/*.h` with all the headers.
For users that don't have both `-Iuhdm/include` and `-Iuhdm/headers` in their include path and use things like `#include "expect_stmt.h"`,  but use full paths such as `#include "uhdm/headers/expect_stmt.h"`, this will be a bit of a disruption to change it to `#include "uhdm/expect_stmt.h"`, but I'd argue that since the user-base is still smallish, this will be less of a problem now than later.

WDYT @alaindargelas @MikePopoloski  @hs-apotell ?

alaindargelas, 2021-09-08:
Fine by me.

MikePopoloski, 2021-09-08:
I've been looking at this and I'm leaning towards not using the generated type system at all and having slang write the capn proto database directly. I think this means I really only need two things from the UHDM repo / model generation process (and therefore don't have much stake in where generated output goes):
1. The UHDM.capnp file, or equivalently the generated .h and .c++ files from it.
2. The uhdm_types.h file, to know the type values consumers will be expecting. I've currently hacked up this file in my local copy to remove everything except the enum.

It would be nice if it was easy to grab these without depending on the whole repo, like as part of some versioned release process.

hzeller, 2021-09-08:
Ok, I think this is an important additional thing we should install: the generated `UHDM.capnp`. In particular a versioned process ties in a bit with https://github.com/chipsalliance/Surelog/issues/1707

Can you open a separate issue on these @MikePopoloski ? There we then can discuss the details.
I also think it might be a good time starting to create releases that contain the created aritfacts.

hs-apotell, 2021-09-08:
Be warned - Issues like these are Pandora's box. They tend to spin out of control very fast.

That said, I agree that the _published_ release should be more organized. There should be the standard _include_, _lib_, and _bin_ folders where headers, static libraries and executables go.

> there is also uhdm.h installed twice

If I remember correctly, I already fixed that with one of my previous commits. Please confirm if this is still an issue.

While we are it, here are few other _not-so-nice_ stuff that needs attention to get to a production ready codebase.

* *Use of Tcl for auto-generation* - However, easy and flexible Tcl might be, it is still not developer friendly. Especially, for developers on Windows where there is no standard installation procedure. Switching to Python or something more mainstream would really help drive more interest.
* *Include headers and their order* - We should follow "include what's needed" and also follow the standard order where all include files are referenced using <> and not using quotes (unless of course something is private implementation and is required only in cpp file in which case the file should not be part of published release). All system headers should go at the end of include list. And, yes, sorted ordering.
* *Location of generated files* - Treat source folders are read-only and move all auto-generated files into output/build folder.
* *Size of generated code* - We have explicitly turned off optimization for certain files and also are forced to use `/bigobj` flag to be able to build source code. The latter is not so bad but former is. Some of the generated files are rather large and should be broken up. This might require some rethinking/redesigning on the auto-generation front. Among the low hanging fruits -

  * Move class specific serialization methods in class specific source files. This will generate more source files (so more modular code) and prevent the Serialization_Save/Restore files from getting too large.

  * Remove code duplication in generated code. Every visit/listen method is flattened. This is not ideal and should be revisited to reduce duplication. There is also an callback ordering problem which needs attention too.

  On windows, the code size is also forcing an increase in default stack size required to be able to execute. This is, unfortunately, a link time parameter and so can't be setup on the library (i.e. AFAIK).

* *Deprecate use of posix commands* - Posix commands are NOT portable across platform. There are std version of most commands in use. This is especially a problem for multithreading because of use of `fork`. At the moment, some of operations which are (and should be) threaded are serial on Windows losing vital performance.

P.S. Some of the above shortcomings might be in UHDM and others in Surelog,

hzeller, 2021-09-08:
Agreed, ordering of the headers and IWYU. Addressed some in #452 #417 and #77 to run clang-format that actually sorts headers is open for a while and if I find cycles, I'll have a look at it.
In Surelog, I cleaned up the inclusion order already, but having a IWYU check on a pull request is still something on my TODO.

Tcl might be hard to change, as Alain is proficient in it :) If generating with something else, I'd actually write it in C++ for simplicity.

Code size generation: I've made some improvements in the past, but there are still some to go. Mostly using the class hierarchy of the generated code instead of switching between enums to generate much more compact code. But not all generated code is doing that yet (example #297)

Generated files location I am addressing in #485 (prepared that in #423 #469 )

.... Aaanyway, this is about the install location, and looks like everyone agrees.

alaindargelas, 2021-09-09:
@hs-apotell, create separate issues for all of the above. Some might take longer than other to get fixed (Like Tcl replacement), but we'll give it a go.

PR #488 (2021-09-12): Install all uhdm related headers in flat $PREFIX/include/uhdm/*.h
**DON'T MERGE INTO SURELOG YET**

This removes the arbitrary distinction between relatively
meaningless names header/ and include/

Adapt all code that used the previous paths in their includes
to point to the new location.

Install location is presumed to be a 'system location', i.e. the distributed includes are used in angled brackets,
such as `#include <uhdm/uhdm.h>`

This likely cause one-time disruption for current users that directly included things from `uhdm/headers/foo.h` as these are now in `uhdm/foo.h`; currently the  [... cut]
commit c226ef21 (2021-09-12): Install all uhdm related headers in flat $PREFIX/include/uhdm/*.h

This removes the arbitrary distinction between relatively
meaningless names header/ and include/

Adapt all code that used the previous paths in their includes
to point to the new location.

Install location is presumed to be a 'system location', i.e.
the includes are used in angled brackets,
such as #include <uhdm/uhdm.h>

This likely cause one-time disruption for current users, but
currently the user-base is small enough to be a limited
disruption.

While at it, install also the UHDM.capnp in $PREFIX/lib/uhdm
for other tools  [... cut]
closed 2021-09-12 by alaindargelas