chipsalliance/UHDM #486: Reduce size of generated classes

hzeller, 2021-09-08:
There is a lot of code that can be compacted by using the natural inheritance of the class hierachy. I've done a bunch last year, looks like it is a good time for another round.

hs-apotell, 2021-09-15:
Could you add more information on what this task entails. Is this in context to https://github.com/chipsalliance/Surelog/issues/1838?

hzeller, 2021-09-15:
An example would be #297.

Mostly the observation was that generated code was getting large in many cases there were switch-case statements generated in  the code, but it was re-doing a lot of things for all the base-classes and then the object in question.
Also often with gigantic switch/case statements which will actually be a runtime issue (branches, non-local code etc.)

So the approach in the above mentioned pull request was to actually use the object hierachy, delegate some work to the super class and only do the remaining work in the final part.
I remember that there ware more things that can benefit from a likewise treatment.

hs-apotell, 2021-09-20:
I intend to do similar treatment as in #297 to visitor and listener logic. I also intend to go a step further and reduce the logic in each implementation by calling its base class implementation.

```
void Base::DeepCopy(Base *const instance) const {
...
}

void Derived::DeepCopy(Derived *const instance) const {
    Base::DeepCopy(instance);
    ....
}

Derived* Derived::DeepClone() const {
    Derived *const instance = MakeInstance();
    DeepCopy(instance);
    return instance;
}
```

hzeller, 2021-09-20:
Sounds good. Making use of the class hierachy will bring a lot of gains.

hs-apotell, 2021-11-12:
@hzeller  #547 resolves this. If there is anything else that you think can be done, please report back.

alaindargelas, 2022-01-01:
This is all merged in.

closed 2022-01-01 by alaindargelas