chipsalliance/UHDM #499: Add a call to release all memory used by UHDM

alaindargelas, 2021-09-14:
After UHDM has been translated to the native application data-structure (Yosys,Verilator....), before invoking the application business logic, release UHDM memory which should not be needed anymore at that point.

alaindargelas, 2021-09-14:
@kbieganski to test in Yosys when this is ready.

hzeller, 2021-09-17:
Probably not at this stage of development as we want first the dust to settle and the Yosys to compile all the SystemVerilog.

But then a possible future idea for users of UHDM serialized data: having a generated lightweight API that has the same interface as read/write UHDM, but only providing read-only methods that then just delegates to the underlying Capnproto data ? The capnproto data can be loaded as block into memory (or even memory `mmap()`-ped on OSes that support that), and released at once.
Since the backing memory is stable, we even can hand out `std::string_views` to avoid any copying and create expensive temporary objects.

alaindargelas, 2021-09-17:
@hzeller, If you feel like it, please open another issue for the idea above.

This issue is closed.
@kbieganski, do call SURELOG::shutdown_compiler function when you are done populating UHDM data into Yosys, prior to invoque Yosys business logic. UHDM will be purged from memory.

closed 2021-09-17 by alaindargelas