chipsalliance/UHDM #510: Add matrix to test different script code generations

hzeller, 2021-09-20:
Currently, there are two scripts for generating the code. Tcl, which is maintained by Alain as it is his language of choice and an experimental Python script that intends to potentially replace that in the future.
We need to make sure the code generation does not diverge while they are both there. So we need

  * [ ] A [matrix](https://docs.github.com/en/actions/learn-github-actions/managing-complex-workflows#using-a-build-matrix) in the CI to run the code generation and tests.
  * [ ] Structural comparison: Testing that the both codes are equivalent is probably hard (different spacing and such), but to show that they are structurally equivalent should be simple: compare the corresponding files and check that they have the same nesting of structure elements  `tr -dc "{}[]()" < some-generated-file.cc`, which then results a string like `(){(){({})(){()}}(){(())[]}(){()(())}}` - compare these 'structure fingerprint'-strings. While this is only a quick smoke test, it will easily detect if there are additional branches etc. in one vs. another.
  * [ ] Also a matrix in Surelog; I think in particular running the regression there with two different backends makes sense.

These tests should be in place before further changes happen in the scripts so that we have the confidence that things are equivalent.

@hs-apotell would you like to take on this task ?

hs-apotell, 2021-09-20:
Our team is actively working on some of the reported issues to reduce code duplication and making the generated code more modular. Our focus is only on the Python version (we don't have a Tcl expert to really pull off this change in other version).

We expect the change to hit PR stage very soon. I don't think either of these tasks would make much sense once this change hits master.

Moreover, one of the most important difference between the two versions is that Python retains the order of declarations in model_def file while Tcl doesn't. That makes the generated code very different - Functions are ordered differently and the code within each function is ordered differently. A simple regex like what you suggest wouldn't work. We spent a lot of time to compare the generated code between two versions to ensure equivalence and all that was manual. Doing that programmatically would be a whole different challenge.

hzeller, 2021-09-20:
At least point 1 and 3 for the CI test be implemented.

alaindargelas, 2022-01-01:
Switch to Python generation completed

closed 2022-01-01 by alaindargelas