chipsalliance/UHDM #556: Ideal way to fetch all vpiStmts

mysoreanoop, 2021-11-04:
Would appreciate some guidance:

We are looking for If/IfElse statements (in procedural assignments), ternary operators (in continuous and procedural), and case statements (procedural?). What would be the ideal way to do this? 

Currently we are doing a recursion of finding all vpiContAssign and vpiProcess, and peeking in recursively traversing most major nodes until we hit statements. There doesn't appear to be a vpi_iterate through statements when in vpiModule scope (is there?). Would it be easy for us to implement that? Or was that not implemented for a reason?

alaindargelas, 2021-11-05:
. There is no vpi_iterate through vpiStmts when in vpiModule scope (is there?).

  No there is not, this is not in the standard VPI.

. Would it be easy for us to implement that? 

 Yes, it should be, if you can make it generic in taking an arbitrary set of object types you want to query, and an arbitrary set of stop objects you want the query not to traverse, it would be higher value for other people to use.
 There is a similar Surelog internal set of routines that operate on the grammar AST: 
https://github.com/chipsalliance/Surelog/blob/8bbae3af233cf8244c666b16b5eb5a0a49e7f2ad/src/Design/FileContent.cpp#L469
  You need to develop (And please contribute it back) a similar one the UHDM data model.

. Or was that not implemented for a reason? 
  No one asked for that

Is there a better way to accomplish this?
  No

mysoreanoop, 2021-11-06:
Got it, thank you. I'll try to do that when I get the time; will definitely contribute it back if I succeed!

closed 2021-11-06 by mysoreanoop