chipsalliance/UHDM #602: ModuleNotFoundError: No module named 'orderedmultidict'

Thomasb81, 2022-01-02:
Compiling Surelog 8fb385dd2a644df7f6c8cd578bb7223cc469bf5d

The build failed on my Ubuntu 21.10:
```
Traceback (most recent call last):
  File "/home/tom/prog/git/Surelog/third_party/UHDM/scripts/generate.py", line 7, in <module>
    import loader
  File "/home/tom/prog/git/Surelog/third_party/UHDM/scripts/loader.py", line 1, in <module>
    import orderedmultidict
ModuleNotFoundError: No module named 'orderedmultidict'
gmake[3]: *** [third_party/UHDM/CMakeFiles/GenerateCode.dir/build.make:403 : third_party/UHDM/generated/src/UHDM.capnp] Erreur 1
```

Apparently my python3 installation does not have this extra package : orderedmultidict

Couldn't we limit to python package coming from standard installation ?

alaindargelas, 2022-01-02:
@hs-apotell please address by documenting.
I had to do the following to be able to do:
- make: pip install orderedmultidict
- sudo make install: sudo pip install orderedmultidict

hs-apotell, 2022-01-04:
> Couldn't we limit to python package coming from standard installation ?

Unfortunately, Python doesn't have a data structure required to implement a recurring predictable output. I tried to make the implementation part of the solution itself by adding the file to the repository but that implementation broke with the new python release. Thinking this is going to be a recurring problem, I decided to use a public community maintained solution to the problem.

alaindargelas, 2022-01-04:
This has been documented and addressed in CI

closed 2022-01-04 by alaindargelas