chipsalliance/UHDM #603: No implementation of vpi_compare_objects function

RRozak, 2022-01-04:
`vpi_compare_objects` function is unimplemented.

I need it to check if in the example below `vpiParent` refers to this `var_select` or some other if there would be more.
I tried to compare the names, but this function will make it much simpler and should work in general.

```
      \_var_select: (work@top.a.a), line:3:20, endln:3:29, parent:work@top.a
        |vpiName:a
        |vpiFullName:work@top.a.a
        |vpiIndex:
        \_constant: , line:3:22, endln:3:23, parent:work@top.a.a
          |vpiDecompile:0
          |vpiSize:64
          |UINT:0
          |vpiConstType:9
        |vpiIndex:
        \_part_select: , parent:work@top.a.a
          |vpiParent:
          \_var_select: (work@top.a.a), line:3:20, endln:3:29, parent:work@top.a
          |vpiConstantSelect:1
          |vpiLeftRange:
          \_constant: , line:3:25, endln:3:26
            |vpiDecompile:3
            |vpiSize:64
            |UINT:3
            |vpiConstType:9
          |vpiRightRange:
          \_constant: , line:3:27, endln:3:28
            |vpiDecompile:0
            |vpiSize:64
            |UINT:0
            |vpiConstType:9
```

alaindargelas, 2022-01-19:
@hs-apotell, that new function like vpi_iterate would benefit from being auto-generated. Can you take on that task?
Since we switched to the Python code generation, I would not know where to start.
Basically for 2 given objects, compare that their trees are isomorphic and values are equal, note, the vpifFullName value should not be checked.

hs-apotell, 2022-01-19:
@alaindargelas Will look into it shortly.

hs-apotell, 2022-01-19:
@alaindargelas @RRozak Few questions about the potential use cases and implementation -
* Is this supposed to be a shallow compare or a deep compare?
* Is the expected result a boolean (true or false) or integer (less, equal, greater) or are we interested in more detailed information on the actual differences themselves?

alaindargelas, 2022-01-20:
It should be a deep compare, only equal not equal.

alaindargelas, 2022-01-20:
Ordered compare, compare in place the vectors

alaindargelas, 2022-02-08:
Fixed by #618

closed 2022-02-08 by alaindargelas