chipsalliance/UHDM #611: Continuous assignment is not supported, only individual assignment is supported

0616ygh, 2022-01-10:
When I use UHDM+Yosys for my Systemverilog RTL project, I find this tool didn't support that:

input variable_type a,b;

and I need to change it to :
i
nput variable_type a;
input variable_type b;

I would like to ask whether the project will improve on this. Thank you very much.

rkapuscik, 2022-01-11:
Hi, fhanks for your interest in this project! I used the following file to test your issue:
```verilog
module top;
  typedef logic variable_type;
  input variable_type a, b;
endmodule
```
Then I see that both of the variables are present in UHDM:
```
  |vpiPort:
  \_port: (a), line:13:25, endln:13:26, parent:work@top
    |vpiName:a
    |vpiDirection:1
    |vpiLowConn:
    \_ref_obj: (work@top.a)
      |vpiName:a
      |vpiFullName:work@top.a
      |vpiActual:
      \_logic_var: (work@top.a), line:13:25, endln:13:26, parent:work@top
  |vpiPort:
  \_port: (b), line:13:28, endln:13:29, parent:work@top
    |vpiName:b
    |vpiDirection:1
    |vpiLowConn:
    \_ref_obj: (work@top.b)
      |vpiName:b
      |vpiFullName:work@top.b
      |vpiActual:
      \_logic_var: (work@top.b), line:13:28, endln:13:29, parent:work@top
```
(I got this by running Surelog with `-d uhdm` option)

Is my example correct? If not, could you provide a minimal, but complete code to test against? If it is, please open an issue in the [Yosys plugin repository](https://github.com/antmicro/yosys-uhdm-plugin-integration/issues) and we'll look at it, as it definitely should work.

rkapuscik, 2022-01-12:
I actually found another example that seems to reproduce the bug:
```verilog
package my_pkg;
    typedef struct {
        logic[2:0] first;
        int second;
    } variable_type;
endpackage
module top import my_pkg::variable_type;
(
    input variable_type a, b
);
endmodule
```
I've opened https://github.com/chipsalliance/Surelog/issues/2425, since this needs to be fixed in Surelog (it is used to obtain the UHDM file).

0616ygh, 2022-01-13:
Thank you for your patience and validation!  Sorry for the late reply. Our lab is trying to parse SystemVerilog using UHDM+Yosys. I will start collecting test cases and Bug reports from our lab on UHDM and provide reports when errors occurred as much as possible. (I was preparing for the exam these two days and didn't reply in time. Sorry for that.)
For this bug, I have an example： It also appears after "import package".
[image]

alaindargelas, 2022-01-18:
Hi @0616ygh,
please file these issues in https://github.com/antmicro/yosys-uhdm-plugin-integration/issues

closed 2022-01-18 by alaindargelas