chipsalliance/UHDM #615: Run some type of function will meet error

little-blue, 2022-01-11:
Just as the title, when the function has  the "output" local variable, It will meet error in uhdm. The error is like this:
[image]
the code is like this 
[image]
The function definition is this 
[image]

When I expand this function where I use this function. The error disappear.

alaindargelas, 2022-01-18:
Please add the example in text, not with a picture, I am not going to retype what is in the picture, I hope you understand :-)
Use 
```
text section 
```

Also the error is not coming from Surelog/UHDM, what executable are you running?
Same thing, please put the error and a bit more messages above the error in text, not picture.

alaindargelas, 2022-01-18:
This error is filed in the wrong repo.

little-blue, 2022-01-18:
Sorry, I didn't have the experience of submitting before, I will abide by the etiquette of submitting later, thank you for your teaching

closed 2022-01-18 by little-blue

alaindargelas, 2022-01-18:
No problem, welcome to Github and UHDM!