chipsalliance/UHDM #626: README.md need updates

Thomasb81, 2022-02-05:
The current README.md reference file that have probably be renamed or does not exist anymore, making the difficult the understanding of information provide in the project documentation.

Here are the list of line on on which at least a file reference is wrong:
https://github.com/chipsalliance/UHDM/blob/7ea4781d48ec78ab37438c3c431269cab5150784/README.md?plain=1#L75
https://github.com/chipsalliance/UHDM/blob/7ea4781d48ec78ab37438c3c431269cab5150784/README.md?plain=1#L81
https://github.com/chipsalliance/UHDM/blob/7ea4781d48ec78ab37438c3c431269cab5150784/README.md?plain=1#L104
https://github.com/chipsalliance/UHDM/blob/7ea4781d48ec78ab37438c3c431269cab5150784/README.md?plain=1#L109 (no link and wrong reference)
https://github.com/chipsalliance/UHDM/blob/7ea4781d48ec78ab37438c3c431269cab5150784/README.md?plain=1#L111
https://github.com/chipsalliance/UHDM/blob/7ea4781d48ec78ab37438c3c431269cab5150784/README.md?plain=1#L112

alaindargelas, 2022-02-07:
@hzeller, I think you changed all of these file names.

alaindargelas, 2022-02-08:
Fixed by #631

closed 2022-02-08 by alaindargelas