chipsalliance/UHDM #628: Debug libuhdm.a size explosion

alaindargelas, 2022-02-07:
After switching to the Python code generation, there has been a drastic increase of the size of libuhdm.a especially the debug version.
I'll suspect the CMakeFiles/uhdm.dir/generated/src/*.o objects (One object per model) that are new after the Python code change.
With the Tcl code generation most of these .o file didn't exist and the library .a was order of magnitude smaller.

alaindargelas, 2022-02-07:
@hs-apotell, please comment on the new objects and the possibility to recombine into single file.

alaindargelas, 2022-02-07:
ls -all dbuild/CMakeFiles/uhdm.dir/generated/src/*.o
-rw-rw-r-- 1 alain alain 12385432 Feb  6 20:24 dbuild/CMakeFiles/uhdm.dir/generated/src/actual_group.cpp.o
-rw-rw-r-- 1 alain alain 12760568 Feb  6 20:24 dbuild/CMakeFiles/uhdm.dir/generated/src/alias_stmt.cpp.o
-rw-rw-r-- 1 alain alain 12784280 Feb  6 20:23 dbuild/CMakeFiles/uhdm.dir/generated/src/always.cpp.o
-rw-rw-r-- 1 alain alain 12780528 Feb  6 20:24 dbuild/CMakeFiles/uhdm.dir/generated/src/any_pattern.cpp.o
-rw-rw-r-- 1 alain alain 13069544 Feb  6 20:25 dbuild/CMakeFiles/uhdm.dir/generated/src/array_net.cpp.o
...

ls -all dbuild/bin/
-rwxrwxr-x 1 alain alain 772488448 Feb  6 20:26 full_elab_test
-rwxrwxr-x 1 alain alain 795683032 Feb  6 20:27 garbage_collect_test
....
 ls -all dbuild/lib/
-rw-rw-r-- 1 alain alain 3497872010 Feb  6 20:26 libuhdm.a    !!!!!!!!!

alaindargelas, 2022-02-07:
Loading the debug executable in gdb has also taken an order of magnitude slower because of the size of the executable.

hs-apotell, 2022-02-07:
Let me investigate.

alaindargelas, 2022-02-10:
@hzeller can you take a look?

alaindargelas, 2022-02-11:
Fixed

closed 2022-02-11 by alaindargelas