chipsalliance/UHDM #752: does uhdm library provide function that directly transform uhdm format to verilog or systemverilog?

tarik-jin, 2022-06-20:
I found uhdm a little bit like LLVM IR, which could be consumed by different kinds of codegen backend, so i wonder if we could output/dump verilog/systemverilog from uhdm after we do some transformation on udhm format.

alaindargelas, 2022-06-22:
@tarik-jin, completely possible.
@hs-apotell has done a "first step" toward that called the rountrip executable in Surelog: https://github.com/chipsalliance/Surelog/blob/master/src/roundtrip.cpp
A "pretty printer" that decompiles UHDM into Verilog (down cast) or SystemVerilog (as it was parsed) is completely possible and would be a welcome contribution.

closed 2022-10-24 by tarik-jin

alaindargelas, 2022-10-24:
@tarik-jin did you implement this? Would you contribute the code?

tarik-jin, 2022-10-24:
sorry, I don't implement this feature, I'd like to, but I don't have enough time.
do I need to reopen?

alaindargelas, 2022-10-24:
No keep it closed. Github said the PR has been closed as completed, that is why I was inquiring.