chipsalliance/UHDM #831: Linking error on Windows with MSYS2 (MINGW64 environment)

RustamC, 2022-12-08:
Hi!

I've tried to build the [systemverilog plugin](https://github.com/chipsalliance/yosys-f4pga-plugins/tree/main/systemverilog-plugin) for yosys on Windows using [msys2/mingw64](https://www.msys2.org/docs/environments/). It links to the [Surelog](https://github.com/chipsalliance/Surelog/) and **UHDM** static libraries. So I built Surelog which also builds UHDM. 

All went well, but during compilation and linking of the plugin I ran into a few problems with Surelog & UHDM.

1) At link time, there were lots of errors like `"ld: unresolved external symbol __imp__vpi_get"`. After a little investigation, I've discovered that this is due to the [include/vpi_user.h ](https://github.com/chipsalliance/UHDM/blob/8fb2ae051cd1ada720d2814e432622afb7b09ca4/include/vpi_user.h) file. For some reason, during compilation with MinGW, the `__declspec(dllimport)/__declspec(dllexport)` keywords are added to function declarations:

https://github.com/chipsalliance/UHDM/blob/8fb2ae051cd1ada720d2814e432622afb7b09ca4/include/vpi_user.h#L66

https://github.com/chipsalliance/UHDM/blob/8fb2ae051cd1ada720d2814e432622afb7b09ca4/include/vpi_user.h#L79

After removing `defined(__MINGW32__) ` the problem was solved. 
Is it possible to delete this defined for being useless?

2) At compile time it turned out that some headers contain identifiers, which are already used in [Windows API headers](https://learn.microsoft.com/en-us/windows/win32/winprog/using-the-windows-headers). For example, the generated `interface.h` contains the definition of `class interface`. And the Windows API already uses `interface`. I solved this problem by adding **-DWIN32_LEAN_AND_MEAN** flag during compilation. Maybe this should be noted in the README?

alaindargelas, 2022-12-13:
@hs-apotell FYI, some other issues for the MSYS2 build to look at.

RustamC, 2022-12-14:
1. Yes, it's true when building UHDM. I've also added these flags to my [custom Yosys makefile](https://github.com/RustamC/yosys/compare/win-static-build-024...RustamC-patch-1) and systemverilog plugin linked successfully. 
2. Thank you, @hs-apotell !

closed 2022-12-28 by alaindargelas