chipsalliance/UHDM #872: `-DDEBUG` allows switching between two incompatible class layouts in a public header file

mglb, 2023-02-28:
[`SymbolId.h`, one of the public headers, uses `DEBUG` preprocessor define][debug-def] to switch between two binary-incompatible class layouts and sizes by optionally adding a data field to a class.
As a result, a program compiled with `-DDEBUG` which uses `SymbolId.h` and UHDM compiled without `-DDEBUG` will crash.

This has been a minor issue in systemverilog-plugin, as Yosys sets `-DDEBUG` when `ENABLE_DEBUG` is set.

Moreover, seems like the `DEBUG` define is not set by CMake when UHDM (or, more precisely, Surelog) is compiled with `-DCMAKE_BUILD_TYPE=Debug`. `-DCMAKE_CXX_FLAGS='-DDEBUG=1'` makes everything work fine.

[debug-def]: https://github.com/chipsalliance/UHDM/blob/master/templates/SymbolId.h#L13

mglb, 2023-02-28:
I wonder whether `#if defined(DEBUG)` can be removed, as the relevant feature can be already controlled with a more unique `SYMBOLID_DEBUG_ENABLED`.

Another option would be to generate SymbolId.h with hardcoded option depending on whether UHDM has been compiled with it or not.

alaindargelas, 2023-02-28:
@hzeller, please suggest.

alaindargelas, 2023-03-02:
@hs-apotell this concerns you.

hzeller, 2023-03-02:
It is highly bug-prone if the layout of objects is changed depending on how the something happens to be compiled.
So I suggest to remove the `#ifdef` sections with debug information in this class. Are there other places like that as well ? Then these also.

If this is still needed for local debugging @hs-apotell , I suggest for now to keep the change in a local patch that you can apply to your environment when needed, but remove it from the version on github to not trip users further.

If the debugging feature is needed outside the core developer circles, we then can still see if we find a less intrusive solution.

PR #876 (2023-03-08): Don't use -DDEBUG to change layout of SymbolId
Since -DDEBUG is commonly used, it is dangerous to change the layout of an object just because of that - the project using UHDM will now accidentally change the size that is not compatible anymore with the library.

(This might not have detected before as UHDM typically was compiled alongside Surelog with the same flags, but 3rd party applications just using UHDM will break).

Fixes #872
commit bc9aa034 (2023-03-08): Don't use -DDEBUG to change layout of SymbolId

Since -DDEBUG is commonly used, it is dangerous to change the
layout of an object just because of that - the project using
UHDM will now accidentally change the size that is not compatible
anymore with the library.

(This might not have detected before as UHDM typically was
compiled alongside Surelog with the same flags, but 3rd party
applications just using UHDM will break).

Fixes #872
closed 2023-03-08 by hzeller