chipsalliance/UHDM #888: Need to run git tagging in CI action - permission denied

hzeller, 2023-03-22:
I try to [tag the revision whose SoT is in the CMake file](https://github.com/chipsalliance/UHDM/blob/master/.github/workflows/main.yml#L266-L293) on push whenever it changes.

However, it looks like we need some additional permissions in the github action.
https://github.com/chipsalliance/UHDM/actions/runs/4495222394/jobs/7908625709

I don't have access to the 'settings' tab in UHDM, which I suspect I need to find the right settings.

Tagging a few people who might know how to achieve this
@mithro @kgugala @umarcor

kgugala, 2023-03-23:
#889 attempts to fix this

hzeller, 2023-03-23:
Looks like we need a bot account; @kgugala fills in details.

For now, I've just manually tagged the release, so that we can explore that permission issue separately.

PR #890 (2023-03-24): Set permissions to write tag.
Fixes: #888
commit 4e211304 (2023-03-24): Set permissions to write tag.

Fixes: #888
closed 2023-03-24 by hzeller