chipsalliance/UHDM #900: [question] Is documentation for UHDM available?

yurivict, 2023-03-31:
For example, ```VpiListener``` has a lot of methods. Is documentation available?

For example, the ```enterPort``` overridable function passes the ```port``` object. object->VpiSize()=0, object->VpiScalar()=0, object->VpiVector()=0. There is no size, it is no a scalar and not a vector, which is difficult to understand.

There should be a documentation that explains this interface and all involved values.

Thomasb81, 2023-03-31:
The idea of this library is to implement Ieee1800.1-2017 chapter 37 VPI object model.
[image]

vpiListener is a class object that will walk through for you through the full object model. So if you are only interest to collect port information you just have to override the `enterPort` method to do your specific stuff around port object.
So method in Listener class provide as many method than object element define by the VPI object model.

I fear that documentating this would be equivalent to repeat what is in the standard.

alaindargelas, 2023-04-01:
@yurivict , not all fields might be filled for all objects in the data model. So for instance, VpiSize is only implemented for typespecs and variables.
So you need to get the typespec for the port and get the size from the typespec.
 
The Boolean VpiScalar is not yet implemented (Only a couple of attempts in Surelog to populate type scalar, but not the bool):
```
~/Surelog$ grep -r -i VpiScalar src/
src/Expression/Value.cpp:      return vpiScalar;
src/Expression/Value.cpp:      return vpiScalar;
src/DesignCompile/CompileHelper.cpp:      c->VpiConstType(vpiScalarVal);
```

Workaround: What is size different than 1 (taken from the typespec) is not a scalar....

A rather large application that uses the UHDM VPI API can be found here:
https://github.com/chipsalliance/yosys-f4pga-plugins/tree/main/systemverilog-plugin
It is the SystemVerilog front-end for Yosys (Following the flow Surelog - UHDM - Yosys-plugin UHDM2AST - Yosys).

The documentation for the VPI is really this file as pointed by @Thomasb81:
https://github.com/chipsalliance/Surelog/blob/master/third_party/Verilog_Object_Model.pdf

closed 2023-04-25 by alaindargelas