chipsalliance/UHDM #901: Add Makefile target for python support testing and some README

alaindargelas, 2023-04-01:
Please add a top Makeflie test target for python:

make test_python

Also add the new target in the CI Action testing:
https://github.com/chipsalliance/UHDM/blob/eef0530d03ab6554ff0c65ab8f2e7e3fbcc5ad9b/.github/workflows/main.yml#L88

Also, please add some documentation in the README

Thomasb81, 2023-04-16:
2 tests are run when UHDM is compile with BUILD_SHARED_LIBS and test are run. (integrated to ctest framework)
https://github.com/chipsalliance/UHDM/blob/04da4ceb9024a10c756ad0f70db38e6ad05d5310/python/CMakeLists.txt#L47
https://github.com/chipsalliance/UHDM/blob/04da4ceb9024a10c756ad0f70db38e6ad05d5310/python/CMakeLists.txt#L52

It seems to me that BUILD_SHARED_LIBS case is already integrated in workflow : https://github.com/chipsalliance/UHDM/blob/eef0530d03ab6554ff0c65ab8f2e7e3fbcc5ad9b/.github/workflows/main.yml#L97

Do we need more ? 
Sorry it is not clear what to do unless if it is in the perspective of https://github.com/chipsalliance/UHDM/issues/911 then it will require a specific entry in the workflow.

alaindargelas, 2023-04-16:
I can't see this running in the Actions, can you point to an Action log that shows this is actually been executed?
I think something is missing here.

Also, please add a section in the README on how one would use this? (Set the compile option, run a  test).

Thomasb81, 2023-04-18:
> But I don't see any tests running.

https://github.com/chipsalliance/UHDM/actions/runs/4737059941/jobs/8409401013?pr=923#step:14:431

This aspect should be ok. Tests are run in the workflow now.

> Also, please add a section in the README on how one would use this? (Set the compile option, run a test).

I add the required paragraph with python code given as example. In some way it contributes to https://github.com/chipsalliance/UHDM/issues/783

closed 2023-04-24 by alaindargelas