chipsalliance/UHDM #911: Python binding should be conditional on a special option, not on BUILD_SHARED_LIBS

yurivict, 2023-04-11:
[This code](https://github.com/chipsalliance/UHDM/blob/master/CMakeLists.txt#L408) makes ```add_subdirectory(python)``` conditional on ```BUILD_SHARED_LIBS```.

I think that this is a mistake.

Version: 1.54

alaindargelas, 2023-04-12:
@Thomasb81 can you take a look at this and #901

yurivict, 2023-04-12:
python binding should be conditional on a special option, like ```option(BUILD_PYTHON_BINDINGS [...])```, or it should ideally be a separate project, because bindings are essentially separate projects that people tend to bundle together with the main project.

Thomasb81, 2023-04-16:
I can probably provide a PR for an option as suggest by @yurivict but not sure to understand the separate project option. 

Are you referring to cmake project or git dependency ?

yurivict, 2023-04-16:
> Are you referring to cmake project or git dependency ?

I am referring to a separate git project for the Python binding.

For example, there is a Heyoka C++ project here: https://github.com/bluescarni/heyoka
And there is it's Python binding here: https://github.com/bluescarni/heyoka.py

If you would look at the Python binding itself - it has nothing to do with the C++ project, except that it links to it like any other dependency would. The Python binding is not a "part" of UHDM: The rest of UHDM does not use the Python binding code. The Python binding is a separate project that actually can be installed multiple times for multiple concurrent Python versions.

Thomasb81, 2023-04-18:
> The Python binding is a separate project that actually can be installed multiple times for multiple concurrent Python versions.

What you say make sense, in particular if this additional project is host by chipsalliance. 
But sorry I don't feel concerned by this need. Also I see in the past weeks so many commit to fine tune the cmake build system between uhdm and surelog ... With my knowledge and understanding of cmake ; that's not a task for me. :)

alaindargelas, 2023-04-25:
Fixed by #924.
To build the python API, please use:
make release-shared UHDM_WITH_PYTHON=ON

closed 2023-04-25 by alaindargelas