chipsalliance/UHDM #1037: Create a RTL Netlist view

alaindargelas, 2023-10-05:
1) Create a Five-box Data Model similar to:
https://www.intel.com/content/www/us/en/docs/programmable/683236/22-4/design-netlist-infrastructure-beta.html
within the UHDM schema, with back pointers to VPI Schema

2) Create a new UHDM application that reads-in UHDM VPI model and write out UHDM Netlist Model along with necessary VPI model

3) Write the code that transforms the UHDM elaborated tree into the Netlist model
    Complex data-type flattening 
    Net Drive-Load graph
    Instance hierarchy
    SysteVerilog Interface flattening
    Optional flattening
    Inference of common elements (DFF....)

alaindargelas, 2023-10-05:
This will create the necessary datastructure to populate Yosys::RTLIL directly (Bypassing Yosys::AST) or other tools necessitating a netlist view of the design.

Thomasb81, 2023-10-06:
> 1. Create a Five-box Data Model

Do you mean a data model suitable for a graphical representation of a design or a part of ?

alaindargelas, 2023-10-06:
The 5 box model is the standard EDA model for modeling a hierarchical netlist.
It is indeed also useful to represent netlist graphically, but that is secondary.

QuantamHD, 2023-10-06:
Cool references @alaindargelas. Sounds like a great idea

alaindargelas, 2023-10-06:
@QuantamHD , I'm currently learning the code of synlig, trying to address some of the cases I opened.
Once I resolve all my outstanding issues, and maybe a couple I saw opened recently, I'll switch to this case.
This project is probably a couple of years out with current staffing, in the meantime we need synlig support.
I'm trying to enroll some help from the community here.

Donmar001, 2023-12-14:
@alaindargelas How to identify the DFF in the code by the VPI,use the vpi_get_str to map the name?

alain-rs, 2023-12-14:
If the dff is instantiated in the input file, then you need to consider it as a blackbox , it has to go through synthesis unaltered, for that you need to read along with your input file another Verilog file containing the blackbox definition of the DFFs. Use the (* blackbox *) attribute, that is already supported in my fork of Synlig, similarly here, the blackbox would pass through and be regurgitated as-is in this netlist view.

If the dff is always statement in the input file, then we need a piece of code called inference that does control/data path analysis and infers the behavior of clock set reset, a similar code exists in Yosys.

Donmar001, 2023-12-15:
> If the dff is instantiated in the input file, then you need to consider it as a blackbox , it has to go through synthesis unaltered, for that you need to read along with your input file another [...]

emmm,i dont really understand it,i just want to extract all the DFF in the netlist,can you list a example to explain it

Donmar001, 2023-12-15:
[tmp.zip](https://github.com/chipsalliance/UHDM/files/13680170/tmp.zip)
@

> If the dff is instantiated in the input file, then you need to consider it as a blackbox , it has to go through synthesis unaltered, for that you need to read along with your input file another [...]

[tmp.zip](https://github.com/chipsalliance/UHDM/files/13680175/tmp.zip)
this is the netlist example

alaindargelas, 2023-12-15:
It looks like you can use a simpler solution with hardcoded names of dffs.
Traverse all the modules recursively (Write a UHDM executable like the example hellouhdm in Surelog).
Use vpi_iterate(vpiModule)  and dump all the ones that match the dff name you are looking for.

Donmar001, 2023-12-15:
> It looks like you can use a simpler solution with hardcoded names of dffs. Traverse all the modules recursively (Write a UHDM executable like the example hellouhdm in Surelog). Use vpi_iterate(vpiModule) and dump all the ones that match the dff name you are looking for.

thk u,bro leave a email to communicate the thechnology.sent message to me bro.my email is  donmarjr148@gmail.com

alaindargelas, 2025-07-27:
This requirement is going to be fulfilled with the project: https://github.com/alainmarcel/uhdm2rtlil
This project will create a Yosys netlist using the RTLIL model pre-synthesis (RTL netlist).
The user can operate at that level of invoke any Yosys Synthesis passes to lower down the representation, at-will, until gate-level.

closed 2025-07-27 by alaindargelas