chipsalliance/UHDM #1070: Port Size Information in UHDM VPI Log Report

Divya2030, 2023-12-22:
While utilizing the UHDM (Universal Hardware Data Model) VPI log report for analyzing Verilog/SystemVerilog designs, I have observed that the report does not directly include port size (width) information, particularly in the `vpiPort` sections for input/output ports or reg/wire netype sections.

 **Current Behavior:** 
- The `vpiPort` section reports the `vpiName` and `vpiDirection` (where `vpiDirection:1` indicates an input port, and other values indicate output or inout ports).
- To determine the size of each port, it is necessary to trace the connected net (`vpiLowConn` or `vpiHighConn`) and assess its typespec or size attribute.
- The type of signal (reg or wire) is indicated by `vpiNetType`, with `vpiNetType:48` for reg and `vpiNetType:1` for wire.

 **Question Regarding Current Approach:** 
- Is identifying input and output ports based on the `vpiDirection` attribute and tracing the connected net for size information the recommended approach in UHDM? Are there more efficient methods available within the UHDM framework for this purpose?
- Any guidance or feedback on best practices for extracting port information using UHDM would also be greatly appreciated.

[surelog_log.zip](https://github.com/chipsalliance/UHDM/files/13752875/surelog_log.zip)

Thomasb81, 2023-12-22:
Hi

```
  |vpiPort:
  \_port: (reset_n), line:3:11, endln:3:18
    |vpiParent:
    \_module_inst: work@i2c_master (work@i2c_master), file:/home/divya/AAA/examples/new_examples_0/i2c_master/i2c_master.sv, line:1:1, endln:79:10
    |vpiName:reset_n
    |vpiDirection:1
    |vpiLowConn:
    \_ref_obj: (work@i2c_master.reset_n.reset_n), line:3:11, endln:3:18
      |vpiParent:
      \_port: (reset_n), line:3:11, endln:3:18
      |vpiName:reset_n
      |vpiFullName:work@i2c_master.reset_n.reset_n
      |vpiActual:
      \_logic_net: (work@i2c_master.reset_n), line:3:11, endln:3:18
    |vpiTypedef:
    \_ref_typespec: (work@i2c_master.reset_n)
      |vpiParent:
      \_port: (reset_n), line:3:11, endln:3:18
      |vpiFullName:work@i2c_master.reset_n
      |vpiActual:
      \_logic_typespec: , line:3:11, endln:3:11
```

As you can read chapter 37.14 of Ieee1800.1-2017 and see in your shared log, `vpiPortType`, `vpiSize`, `vpiScalar` and `vpiVector` are missing. Probably Surelog need to be improve to properly fill those attribute in UHDM.

There are those issue  : 
https://github.com/chipsalliance/UHDM/issues/1037
https://github.com/chipsalliance/Surelog/issues/1565

That discuss more generally the "connectivity feature". I understand that they remain "to be develop".

So probably there could be more efficient way, in term of UHDM database browsing but they rely on not present data...

alaindargelas, 2024-01-05:
@Divya2030 , to your question:
"Is identifying input and output ports based on the vpiDirection attribute and tracing the connected net for size information the recommended approach in UHDM? Are there more efficient methods available within the UHDM framework for this purpose?"

Correct, the net with the same name as the port is the one carrying the size information in the form a the typespec.

The method int ExprEval::size(typespec* tps...) in UHDM will return the size the net, no matter how complex is the typespec.

closed 2024-01-05 by alaindargelas