chipsalliance/UHDM #1074: bindAny should not look in the current instance for HighConn, only parent instance.

pieter3d, 2024-02-07:
See
https://github.com/chipsalliance/Surelog/issues/3958#issuecomment-1931459568

In ElaboratorListener.cpp when bindAny is called and it comes from a port HighConn in VpiListener.cpp, searching for the net by name should not start at the top of the stack. Doing so will cause the HighConn to incorrectly reference the port net inside the module, if the net connected to the port (the desired high conn) has the same name as the port itself.

Assuming I'm not missing something here, this seems like a real bug and makes me wonder how this was never seen before.

alaindargelas, 2024-02-09:
The instStack_ has to be popped temporarily while binding high_conn so is skips the current module scope.

alaindargelas, 2025-06-13:
Fixed by #1110

closed 2025-06-13 by alaindargelas