chipsalliance/UHDM #1094: `$size` evaluates to the wrong value

Rodrigodd, 2025-02-08:
## Description

I was trying to get a existing design loaded into [Synlig](https://github.com/chipsalliance/synlig), but I notice that `$size` of a unpacked array was returning the size in bits of the full memory, instead of just the number of elements.

I am not sure if I should have filled this issue on [Synlig](https://github.com/chipsalliance/synlig), [Surelog](https://github.com/chipsalliance/surelog) or here, but, from what I can tell, the code responsible for evaluating `$size` is mostly here.

## Reproduction

If I have the file `testsize.sv` below:

```systemverilog
module top();
	logic [3:0] a[1:0];
	logic b[$size(a)-1:0];
endmodule
```

And in the Surelog repository I run:

```
dbuild/bin/surelog -sv testsize.sv -top top -parse -nobuiltin -v4 -d uhdm
```

I can see from the output that the `vpiSize` of `b` is 8, where it should be 2.

```
\_module_inst: work@top (work@top), file:/home/rodrigodd/repos/Surelog/testsize.sv, line:1:1, endln:4:10
  |vpiName:work@top
  |vpiVariables:
  \_array_var: (work@top.a), line:2:14, endln:2:15
    |vpiSize:2
  \_array_var: (work@top.b), line:3:8, endln:3:9
    |vpiSize:8
```

<details>
<summary>Full Output</summary>

```
[INF:CM0023] Creating log file "/home/rodrigodd/repos/Surelog/slpp_all/surelog.log".
[WRN:CM0010] Command line argument "-v4" ignored.
[WRN:PA0205] /home/rodrigodd/repos/Surelog/testsize.sv:1:1: No timescale set for "top".
[INF:CP0300] Compilation...
[INF:CP0303] /home/rodrigodd/repos/Surelog/testsize.sv:1:1: Compile module "work@top".
[INF:EL0526] Design Elaboration...
[NTE:EL0503] /home/rodrigodd/repos/Surelog/testsize.sv:1:1: Top level module "work@top".
[NTE:EL0508] Nb Top level modules: 1.
[NTE:EL0509] Max instance depth: 1.
[NTE:EL0510] Nb instances: 1.
[NTE:EL0511] Nb leaf instances: 1.
[INF:UH0706] Creating UHDM Model...
[INF:UH0708] Writing UHDM DB: /home/rodrigodd/repos/Surelog/slpp_all/surelog.uhdm ...
[INF:UH0709] Writing UHDM Html Coverage: /home/rodrigodd/repos/Surelog/slpp_all/checker/surelog.chk.html ...
[INF:UH0710] Loading UHDM DB: /home/rodrigodd/repos/Surelog/slpp_all/surelog.uhdm ...
[INF:UH0711] Decompiling UHDM...
====== UHDM =======
design: (work@top)
|vpiName:work@top
|uhdmallModules:
\_module_inst: work@top (work@top), file:/home/rodrigodd/repos/Surelog/testsize.sv, line:1:1, endln:4:10
  |vpiParent:
  \_design: (work@top)
  |vpiFullName:work@top
  |vpiDefName:work@top
  |vpiNet:
  \_logic_net: (work@top.a), line:2:14, endln:2:15
    |vpiParent:
    \_module_inst: work@top (work@top), file:/home/rodrigodd/repos/Surelog/testsize.sv, line:1:1, endln:4:10
    |vpiTypespec:
    \_ref_typespec: (work@top.a)
      |vpiParent:
      \_logic_net: (work@top.a), line:2:14, endln:2:15
      |vpiFullName:work@top.a
      |vpiActual:
      \_logic_typespec: , line:2:2, endln:2:20
    |vpiName:a
    |vpiFullName:work@top.a
    |vpiNetType:36
  |vpiNet:
  \_logic_net: (work@top.b), line:3:8, endln:3:9
    |vpiParent:
    \_module_inst: work@top (work@top), file:/home/rodrigodd/repos/Surelog/testsize.sv, line:1:1, endln:4:10
    |vpiTypespec:
    \_ref_typespec: (work@top.b)
      |vpiParent:
      \_logic_net: (work@top.b), line:3:8, endln:3:9
      |vpiFullName:work@top.b
      |vpiActual:
      \_logic_typespec: , line:3:2, endln:3:23
    |vpiName:b
    |vpiFullName:work@top.b
    |vpiNetType:36
|uhdmtopModules:
\_module_inst: work@top (work@top), file:/home/rodrigodd/repos/Surelog/testsize.sv, line:1:1, endln:4:10
  |vpiName:work@top
  |vpiVariables:
  \_array_var: (work@top.a), line:2:14, endln:2:15
    |vpiParent:
    \_module_inst: work@top (work@top), file:/home/rodrigodd/repos/Surelog/testsize.sv, line:1:1, endln:4:10
    |vpiSize:2
    |vpiTypespec:
    \_ref_typespec: (work@top.a)
      |vpiParent:
      \_array_var: (work@top.a), line:2:14, endln:2:15
      |vpiFullName:work@top.a
      |vpiActual:
      \_array_typespec: 
    |vpiName:a
    |vpiFullName:work@top.a
    |vpiRandType:1
    |vpiVisibility:1
    |vpiArrayType:1
    |vpiRange:
    \_range: , line:2:15, endln:2:20
      |vpiParent:
      \_array_var: (work@top.a), line:2:14, endln:2:15
      |vpiLeftRange:
      \_constant: , line:2:16, endln:2:17
        |vpiParent:
        \_range: , line:2:15, endln:2:20
        |vpiDecompile:1
        |vpiSize:64
        |UINT:1
        |vpiConstType:9
      |vpiRightRange:
      \_constant: , line:2:18, endln:2:19
        |vpiParent:
        \_range: , line:2:15, endln:2:20
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0
        |vpiConstType:9
    |vpiReg:
    \_logic_var: (work@top.a), line:2:14, endln:2:15
      |vpiParent:
      \_array_var: (work@top.a), line:2:14, endln:2:15
      |vpiTypespec:
      \_ref_typespec: (work@top.a)
        |vpiParent:
        \_logic_var: (work@top.a), line:2:14, endln:2:15
        |vpiFullName:work@top.a
        |vpiActual:
        \_logic_typespec: , line:2:2, endln:2:13
      |vpiFullName:work@top.a
  |vpiVariables:
  \_array_var: (work@top.b), line:3:8, endln:3:9
    |vpiParent:
    \_module_inst: work@top (work@top), file:/home/rodrigodd/repos/Surelog/testsize.sv, line:1:1, endln:4:10
    |vpiSize:8
    |vpiTypespec:
    \_ref_typespec: (work@top.b)
      |vpiParent:
      \_array_var: (work@top.b), line:3:8, endln:3:9
      |vpiFullName:work@top.b
      |vpiActual:
      \_array_typespec: 
    |vpiName:b
    |vpiFullName:work@top.b
    |vpiRandType:1
    |vpiVisibility:1
    |vpiArrayType:1
    |vpiRange:
    \_range: , line:3:9, endln:3:23
      |vpiParent:
      \_array_var: (work@top.b), line:3:8, endln:3:9
      |vpiLeftRange:
      \_constant: , line:3:10, endln:3:18
        |vpiParent:
        \_range: , line:3:9, endln:3:23
        |vpiDecompile:7
        |vpiSize:64
        |INT:7
        |vpiConstType:7
      |vpiRightRange:
      \_constant: , line:3:21, endln:3:22
        |vpiParent:
        \_range: , line:3:9, endln:3:23
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0
        |vpiConstType:9
    |vpiReg:
    \_logic_var: (work@top.b), line:3:8, endln:3:9
      |vpiParent:
      \_array_var: (work@top.b), line:3:8, endln:3:9
      |vpiTypespec:
      \_ref_typespec: (work@top.b)
        |vpiParent:
        \_logic_var: (work@top.b), line:3:8, endln:3:9
        |vpiFullName:work@top.b
        |vpiActual:
        \_logic_typespec: , line:3:2, endln:3:7
      |vpiFullName:work@top.b
  |vpiDefName:work@top
  |vpiTop:1
  |vpiTopModule:1
\_weaklyReferenced:
\_logic_typespec: , line:2:2, endln:2:13
  |vpiParent:
  \_logic_var: (work@top.a), line:2:14, endln:2:15
  |vpiRange:
  \_range: , line:2:8, endln:2:13
    |vpiParent:
    \_logic_typespec: , line:2:2, endln:2:13
    |vpiLeftRange:
    \_constant: , line:2:9, endln:2:10
      |vpiParent:
      \_range: , line:2:8, endln:2:13
      |vpiDecompile:3
      |vpiSize:64
      |UINT:3
      |vpiConstType:9
    |vpiRightRange:
    \_constant: , line:2:11, endln:2:12
      |vpiParent:
      \_range: , line:2:8, endln:2:13
      |vpiDecompile:0
      |vpiSize:64
      |UINT:0
      |vpiConstType:9
\_array_typespec: 
\_logic_typespec: , line:3:2, endln:3:7
  |vpiParent:
  \_logic_var: (work@top.b), line:3:8, endln:3:9
\_array_typespec: 
\_logic_typespec: , line:2:2, endln:2:20
  |vpiRange:
  \_range: , line:2:8, endln:2:13
    |vpiParent:
    \_logic_typespec: , line:2:2, endln:2:20
    |vpiLeftRange:
    \_constant: , line:2:9, endln:2:10
      |vpiParent:
      \_range: , line:2:8, endln:2:13
      |vpiDecompile:3
      |vpiSize:64
      |UINT:3
      |vpiConstType:9
    |vpiRightRange:
    \_constant: , line:2:11, endln:2:12
      |vpiParent:
      \_range: , line:2:8, endln:2:13
      |vpiDecompile:0
      |vpiSize:64
      |UINT:0
      |vpiConstType:9
\_logic_typespec: , line:3:2, endln:3:23
===================
[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 2
[   NOTE] : 5
```

</details>

## Fix?

The patch below makes `$size` return the expected value in the case above, but I not sure if it will break others results, or if the function will still be wrong for other types.

```diff
diff --git a/templates/ExprEval.cpp b/templates/ExprEval.cpp
index 51ef9b3..171dc0e 100644
--- a/templates/ExprEval.cpp
+++ b/templates/ExprEval.cpp
@@ -1272,7 +1272,9 @@ uint64_t ExprEval::size(const any *ts, bool &invalidValue, const any *inst,
     case UHDM_OBJECT_TYPE::uhdmarray_typespec: {
       array_typespec *lts = (array_typespec *)ts;
       ranges = lts->Ranges();
-      if (const ref_typespec *rt = lts->Elem_typespec()) {
+      if (!full) {
+        bits = 1;
+      } else if (const ref_typespec *rt = lts->Elem_typespec()) {
         bits = size(rt->Actual_typespec(), invalidValue, inst, pexpr, full);
       }
       break;
```

alaindargelas, 2025-02-26:
@Rodrigodd, please checkout Surelog, apply your patch in the UHDM submodule,
then, 
cd Surelog> python3 tests/regression.py

Check the diffs, 

You can do python3 tests/regression.py update 
then
git diff 

Let me know the results.

Rodrigodd, 2025-02-27:
I tried running `python3 scripts/regression.py run` followed by `python3 scripts/regression.py update`, but that resulted in all `.log` files in `tests/` to became empty (I guess I had to compile the project before running it).

I run `make regression`, which produce no diff (log output: [make-regression.txt](https://github.com/user-attachments/files/19016616/make-regression.txt)).

I then run `make test` (no diff again):

<details>
<summary>Results</summary>

```
Results:
  +--------------------------------------+--------+-------+--------+-------+---------+--------+-------+----------+---------+---------+-----------+
  | TESTNAME                             | STATUS | FATAL | SYNTAX | ERROR | WARNING |   NOTE |  LINT | CPU-TIME | VTL-MEM | PHY-MEM | ROUNDTRIP |
  +--------------------------------------+--------+-------+--------+-------+---------+--------+-------+----------+---------+---------+-----------+
  | 1364_2005                            | PASS   |     0 |      0 |     0 |       1 |    364 |     0 |     0.00 |       0 |       0 |      5/11 |
  | 3SigsSensList                        | PASS   |     0 |      0 |     2 |       0 |    500 |     2 |     0.01 |      36 |      22 |     20/36 |
  | AaFirstTest                          | PASS   |     0 |      0 |     0 |       4 |      5 |     0 |     0.00 |      31 |      14 |       2/7 |
  | AllBinding                           | PASS   |     0 |      0 |     0 |       1 |    170 |     0 |     0.00 |      31 |      14 |      2/10 |
  | AllPackageSignal                     | PASS   |     0 |      0 |     0 |       2 |   1988 |     0 |     0.00 |      31 |      15 |       0/0 |
  | AlwaysNoElab                         | PASS   |     0 |      0 |     0 |       1 |   1097 |     0 |     0.00 |      32 |      16 |      4/12 |
  | AmiqEth                              | PASS   |     0 |      0 |     3 |      27 |      7 | 28854 |    11.78 |     842 |     799 |       0/0 |
  | AmiqSimpleTestSuite                  | PASS   |     0 |      0 |     2 |      10 |     10 | 18893 |     7.82 |     438 |     427 |       0/0 |
  | ApbSlave                             | PASS   |     0 |      0 |     0 |       0 |      7 |    82 |     0.50 |      57 |      42 |       0/0 |
  | Ariane                               | PASS   |     0 |      0 |     0 |      19 |     55 |     0 |    13.83 |    1159 |    1122 |       0/0 |
  | ArianeElab                           | PASS   |     0 |      0 |     0 |       8 | 164841 |     8 |     2.52 |     110 |      89 |       0/0 |
  | ArianeElab2                          | PASS   |     0 |      0 |     0 |       9 | 177224 |     8 |     2.76 |     209 |     200 |       0/0 |
  | ArrayExprFuncArg                     | PASS   |     0 |      0 |     0 |       2 |   1673 |     0 |     0.00 |      30 |      11 |       0/0 |
  | ArrayInst                            | PASS   |     0 |      0 |     0 |       3 |    823 |     0 |     0.00 |      30 |      11 |      1/10 |
  | ArrayMethodIterator                  | PASS   |     0 |      0 |     0 |       2 |    594 |     4 |     0.00 |      35 |      20 |     14/31 |
  | ArrayNet                             | PASS   |     0 |      0 |     0 |       1 |    172 |     0 |     0.00 |      35 |      19 |       1/7 |
  | ArrayTypespec                        | PASS   |     0 |      0 |     0 |       1 |    277 |     0 |     0.00 |      31 |      13 |       0/3 |
  | ArrayVarName                         | PASS   |     0 |      0 |     0 |       1 |    428 |     0 |     0.00 |      31 |      16 |       0/0 |
  | Assert                               | PASS   |     0 |      0 |     0 |       1 |     70 |     2 |     0.00 |      34 |      18 |       1/7 |
  | AssertDelayError                     | PASS   |     0 |      0 |     0 |       3 |   2907 |    30 |     0.00 |      31 |      18 |       0/0 |
  | Assertions                           | PASS   |     0 |      0 |     0 |       3 |    787 |     5 |     0.00 |      31 |      15 |       0/0 |
  | AssertTempError                      | PASS   |     0 |      0 |     0 |       3 |   4834 |    24 |     0.01 |      36 |      20 |       0/0 |
  | AssignmentPatternInAssignmentPattern | PASS   |     0 |      0 |     0 |       4 |   4598 |     0 |     0.00 |      31 |      15 |     10/44 |
  | Assignments                          | PASS   |     0 |      0 |     3 |       1 |   1770 |    13 |     0.00 |      31 |      16 |       0/0 |
  | AssignPattern                        | PASS   |     0 |      0 |     0 |       1 |    149 |     0 |     0.00 |      33 |      18 |       1/4 |
  | AssignPatternArray                   | PASS   |     0 |      0 |     0 |       4 |   5731 |     0 |     0.00 |      31 |      15 |     15/50 |
  | AssignPlus                           | PASS   |     0 |      0 |     0 |       1 |   2127 |     0 |     0.00 |      29 |       5 |       0/0 |
  | AssignRhsFlat                        | PASS   |     0 |      0 |     0 |       1 |    405 |     0 |     0.00 |      31 |      15 |       0/0 |
  | AssignSubs                           | PASS   |     0 |      0 |     0 |       2 |    536 |     0 |     0.00 |      34 |      18 |      4/19 |
  | AssociativeArray                     | PASS   |     0 |      0 |     0 |       1 |    257 |     0 |     0.00 |      32 |      16 |       2/7 |
  | AssumeProp                           | PASS   |     0 |      0 |     1 |       1 |    232 |     2 |     0.00 |      33 |      18 |       0/0 |
[... cut ...]
+make[2]: Leaving directory '${SURELOG_DIR}/third_party/tests/ariane/work-ver'
+make[1]: *** [Makefile:634: verilate] Error 2
+make[1]: Leaving directory '${SURELOG_DIR}/third_party/tests/ariane'
diff --git a/third_party/tests/oh/BasicOh.log b/third_party/tests/oh/BasicOh.log
index 3e84f37b..386ac8b3 100644
--- a/third_party/tests/oh/BasicOh.log
+++ b/third_party/tests/oh/BasicOh.log
@@ -1,282 +1,282 @@
 [INF:CM0023] Creating log file "${SURELOG_DIR}/build/regression/BasicOh/slpp_all/surelog.log".
 [INF:PP0122] Preprocessing source file "$

alaindargelas, 2025-02-28:
@Rodrigodd , please submit a PR here with the code change.
I'll accept it and merge it.
When it is merged, please submit a PR in Surelog with an update of the UHDM submodule, plus a new test that exhibit the feature in Surelog/tests/

Rodrigodd, 2025-03-07:
I was not satisfied with the fix I proposed, so I took a better look to see if I could find a more proper fix for it. From the Verilog spec:

> $size shall return the number of elements in the dimension, which is equivalent to: $high – $low + 1.

So I tried to look for the code that calculates `$high` and `$low` in UHDM. It was a lot more straightforward, so I tried to test the following patch:

```diff
diff --git a/templates/ExprEval.cpp b/templates/ExprEval.cpp
index 171dc0e..bdc4c03 100644
--- a/templates/ExprEval.cpp
+++ b/templates/ExprEval.cpp
@@ -4015,8 +4015,8 @@ expr *ExprEval::reduceExpr(const any *result, bool &invalidValue,
             tps = tp;
           }
 
-          if ((name == "$high") || (name == "$low") || (name == "$left") ||
-              (name == "$right")) {
+          if ((name == "$size") || (name == "$high") || (name == "$low") ||
+              (name == "$left") || (name == "$right")) {
             VectorOfrange *ranges = nullptr;
             if (tps) {
               switch (tps->UhdmType()) {
@@ -4075,20 +4075,33 @@ expr *ExprEval::reduceExpr(const any *result, bool &invalidValue,
                 } else {
                   return lr;
                 }
+              } else if (name == "$size") {
+                // $size(a) == $high(a) - $low(a) + 1
+                if (lrv > rrv) {
+                  bits = lrv - rrv + 1;
+                } else {
+                  bits = rrv - lrv + 1;
+                }
+                found = true;
               }
             }
           }
+          if (!found) {
+            if (name == "$right" || name == "$low") {
+              bits = 0;
+            }
 
-          if (tps) {
-            bits += size(tps, invalidValue, inst, pexpr, (name != "$size"));
-            found = true;
-          } else {
-            if (object) {
-              bits +=
-                  size(object, invalidValue, inst, pexpr, (name != "$size"));
+            if (tps) {
+              bits = size(tps, invalidValue, inst, pexpr, (name != "$size"));
               found = true;
             } else {
-              invalidValue = true;
+              if (object) {
+                bits =
+                    size(object, invalidValue, inst, pexpr, (name != "$size"));
+                found = true;
+              } else {
+                invalidValue = true;
+              }
             }
           }
         } else if (argtype == UHDM_OBJECT_TYPE::uhdmoperation) {
```

But when I tested it, I discovered that Surelog actually doesn't use the code above. Looking a bit more, I found that Surelog also implements it in a very similar way. So I tried the code below:

```diff
diff --git a/src/DesignCompile/CompileExpression.cpp b/src/DesignCompile/CompileExpression.cpp
index d04b498e..cfb5db56 100644
--- a/src/DesignCompile/CompileExpression.cpp
+++ b/src/DesignCompile/CompileExpression.cpp
@@ -2146,13 +2146,8 @@ UHDM::any *CompileHelper::compileExpression(
             result =
                 compileBits(component, fC, List_of_arguments, compileDesign,
                             reduce, pexpr, instance, false, muteErrors);
-          } else if (name == "$size") {
-            NodeId List_of_arguments = fC->Sibling(child);
-            result =
-                compileBits(component, fC, List_of_arguments, compileDesign,
-                            reduce, pexpr, instance, true, muteErrors);
           } else if (name == "$high" || name == "$low" || name == "$left" ||
-                     name == "$right") {
+                     name == "$right" || name == "$size") {
             NodeId List_of_arguments = fC->Sibling(child);
             result =
                 compileBound(component, fC, List_of_arguments, compileDesign,
@@ -4666,6 +4661,15 @@ UHDM::any *CompileHelper::compileBound(
         } else {
           return lr;
         }
+      } else if (name == "size") {
+        // size = high - low + 1
+        UHDM::constant *c = s.MakeConstant();
+        int64_t size = (lrv > rrv) ? lrv - rrv + 1 : rrv - lrv + 1;
+        c->VpiValue("UINT:" + std::to_string(size));
+        c->VpiDecompile(std::to_string(size));
+        c->VpiConstType(vpiUIntConst);
+        c->VpiSize(64);
+        return c;
       }
     }
   }
```

Again, to my surprise, I saw that Surelog doesn't reach the point of successfully evaluating `$size`, `$high`, `$low`, etc. From what I have seen, `operand->Typespec()` always returns null:

https://github.com/chipsalliance/Surelog/blob/d9e6771272b6b35c1e4c2cadc008cc2ecbe3cd22/src/DesignCompile/CompileExpression.cpp#L4608-L4611

In the `compileSize` function, there is some code that also appears to get the typespec of an expression, but it is much more complicated. I didn't attempt to port it over.

And looking now, I notice the above is also missing the `$` when comparing the name.

Then I tried to just call `reduceExpr` directly, but I didn't get too far (I also didn't try that much).

In the end, I gave up. I will just submit the ad-hoc solution I presented in the OP.