chipsalliance/UHDM #1101: VPI: Unable to get driver/load

heyfey, 2025-04-10:
Hi, I'm doing design analysis using VPI but cannot get driver/load with vpiDriver/vpiLoad. Does UHDM plan to support this? Or can I find driver/load using other API? 

Thanks a lot!

test.v:
```
module top;
    wire a;
    wire b;
    assign b = a;
endmodule
```

.uhdm created with `surelog -parse -d uhdm -elabuhdm test.v`
[surelog.uhdm.tar.gz](https://github.com/user-attachments/files/19686439/surelog.uhdm.tar.gz)

main.cpp:
```
g++ -std=c++17 main.cpp -I$UHDM_DIR/build/generated $UHDM_DIR/build/lib/libuhdm.a $UHDM_DIR/build/third_party/capnproto/c++/src/capnp/libcapnp.a $UHDM_DIR/build/third_party/capnproto/c++/src/kj/libkj.a -ldl -lutil -lm -lrt -lpthread
```

```
// UHDM
#include <uhdm/uhdm.h>
#include <uhdm/vpi_user.h>

#include <iostream>

int main(int argc, const char** argv) {
  UHDM::Serializer serializer;
  vpiHandle design = nullptr;

  const std::string filename = "./surelog.uhdm";
  const std::vector<vpiHandle>& restoredDesigns = serializer.Restore(filename);
  if (restoredDesigns.empty()) {
    std::cerr << "Failed to restore design from " << filename << std::endl;
    return 1;
  }
  design = restoredDesigns[0];  // consider only the first design

  vpiHandle instItr = vpi_iterate(UHDM::uhdmtopModules, design);
  while (vpiHandle inst_h = vpi_scan(instItr)) {
    vpiHandle iter = vpi_iterate(vpiNet, inst_h);
    while (vpiHandle obj_h = vpi_scan(iter)) {
      char* name = vpi_get_str(vpiFullName, obj_h);
      std::cout << "Net: " << name << ", file:" << vpi_get_str(vpiFile, obj_h)
                << ", line:" << std::to_string(vpi_get(vpiLineNo, obj_h)) << ":"
                << std::to_string(vpi_get(vpiColumnNo, obj_h)) << std::endl;

      vpiHandle driversItr = vpi_iterate(vpiDriver, obj_h);
      if (driversItr == NULL) {
        std::cout << "    Error: driver not found for: " << name << std::endl;
      }
      vpi_release_handle(driversItr);

      vpiHandle loadItr = vpi_iterate(vpiLoad, obj_h);
      if (loadItr == NULL) {
        std::cout << "    Error: load not found for: " << name << std::endl;
      }
      vpi_release_handle(loadItr);
    }
  }
  vpi_release_handle(instItr);
}
```

Output:
```
Net: work@top.a, file:/home/heyfey/git-repos/uhdm-test/verilog/driverTest/test.v, line:2:10
    Error: driver not found for: work@top.a
    Error: load not found for: work@top.a
Net: work@top.b, file:/home/heyfey/git-repos/uhdm-test/verilog/driverTest/test.v, line:3:10
    Error: driver not found for: work@top.b
    Error: load not found for: work@top.b
```

heyfey, 2025-04-11:
@Thomasb81 this should be implemented in Surelog, right? Do you have an idea on how to do it?

Thomasb81, 2025-04-11:
My comment was may be not accurate; I was referring exactly to this comment : https://github.com/chipsalliance/Surelog/issues/1565#issuecomment-888835240

I guess net_load and net_drive data has to be retrieve from lowConn and highConn analyze.
Probably this could be seen as an elaboration step, that does not need any modification in Surelog.

@alaindargelas your comment would be welcome.

heyfey, 2025-04-14:
@alaindargelas does the information in the (elaborated) UHDM sufficient to support vpiDriver/vpiLoad?

alaindargelas, 2025-04-16:
@heyfey @Thomasb81 , correct, this needs to be done in UHDM as a new pass like https://github.com/chipsalliance/UHDM/blob/master/templates/SynthSubset.cpp
It just needs to be invoked in Surelog like SynthSubset is.
The pass can be written as a Listener that populates some intermediate data structure, then a post action after listening that populate the UHDM data with the complete driver/load info