chipsalliance/UHDM #1103: Getting 'None' in python UHDM parser

dpetrisko, 2025-05-01:
Hello! I am trying to explore the python API for UHDM.

$ surelog --version
VERSION: 1.84
BUILT  : Apr 30 2025

First, I create the uhdm database with:
```
$ surelog -sverilog -fileunit -parse -elabuhdm -verbose -nowarning --top-module bsg_chip <sources>
...
[NTE:EL0508] Nb Top level modules: 1.
[NTE:EL0509] Max instance depth: 9.
[NTE:EL0510] Nb instances: 54.
[NTE:EL0511] Nb leaf instances: 21.
[INF:UH0706] Creating UHDM Model...
[INF:UH0707] Elaborating UHDM...
[INF:UH0708] Writing UHDM DB: /home/petrisko/scratch/oss-cad-playground/bsg_pearls/bsg_clk_gen_pearl/test/surelog/slpp_unit/surelog.uhdm ...
[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 0
[   NOTE] : 26
```

This seems to have populated the DB correctly:
```
$ uhdm-hier slpp_unit/surelog.uhdm 
Design name: work@bsg_chip
Instance tree:
work@bsg_chip
work@bsg_chip.dut
work@bsg_chip.dut.btm
work@bsg_chip.dut.btm.bccu
work@bsg_chip.dut.btm.rof[0]
work@bsg_chip.dut.btm.rof[1]
work@bsg_chip.dut.btm.rof[2]
work@bsg_chip.dut.btm.rof[3]
work@bsg_chip.dut.btm.rof[4]
work@bsg_chip.dut.btm.rof[5]
...
```

Based off the README, I have this script, call it test.py:
```
import uhdm

#build uhdm Serializer object
s = uhdm.Serializer()
#Read and uhdm database
data = s.Restore('slpp_unit/surelog.uhdm')

#Your specific application
#Here create an iterator on all module in the first design
module_iterator = uhdm.vpi_iterate(uhdm.uhdmallModules,data[0])

#iterate on all module
while(True):
   vpiObj_module = uhdm.vpi_scan(module_iterator)
   if vpiObj_module is None:
       break
   #print the string pointed by vpiName attribut of vpiObj_module
   print(uhdm.vpi_get_str(uhdm.vpiName,vpiObj_module))

```

This results in this following output:
```
$ python test.py 
None
None
None
None
None
None
None
None
None
None
None
None
None
...
```

Does this process look right or am I missing something big? What can I do to debug? Happy to go in the weeds if you can point me in the right direction. Thanks!

Thomasb81, 2025-05-01:
Odd, I feel this has worked one day ...

top.sv:
```
module top();

A A_i();

endmodule

module A();

B B_i();

endmodule

module B();
endmodule
```

You see that if we add udhm debug message with `-d uhdm`, we have a dump of the surelog.uhdm
```
$surelog -sverilog -fileunit -parse -elabuhdm -verbose -nowarning -d uhdm -nobuiltin  --top-module top top.sv
[...]
|uhdmallModules:
\_module_inst: work@A (work@A), file:/home/tom/prog/git/Surelog/mytests/top.sv, line:7:1, endln:11:10
  |vpiParent:
  \_design: (work@top)
  |vpiFullName:work@A
  |vpiDefName:work@A
  |vpiRefModule:
  \_ref_module: work@B (B_i), line:9:3, endln:9:6
    |vpiParent:
    \_module_inst: work@A (work@A), file:/home/tom/prog/git/Surelog/mytests/top.sv, line:7:1, endln:11:10
    |vpiName:B_i
    |vpiDefName:work@B
    |vpiActual:
    \_module_inst: work@B (work@B), file:/home/tom/prog/git/Surelog/mytests/top.sv, line:13:1, endln:14:10
|uhdmallModules:
\_module_inst: work@B (work@B), file:/home/tom/prog/git/Surelog/mytests/top.sv, line:13:1, endln:14:10
  |vpiParent:
  \_design: (work@top)
  |vpiFullName:work@B
  |vpiDefName:work@B
|uhdmallModules:
\_module_inst: work@top (work@top), file:/home/tom/prog/git/Surelog/mytests/top.sv, line:1:1, endln:5:10
  |vpiParent:
  \_design: (work@top)
  |vpiFullName:work@top
  |vpiDefName:work@top
  |vpiRefModule:
  \_ref_module: work@A (A_i), line:3:3, endln:3:6
    |vpiParent:
    \_module_inst: work@top (work@top), file:/home/tom/prog/git/Surelog/mytests/top.sv, line:1:1, endln:5:10
    |vpiName:A_i
    |vpiDefName:work@A
    |vpiActual:
    \_module_inst: work@A (work@A), file:/home/tom/prog/git/Surelog/mytests/top.sv, line:7:1, endln:11:10
|uhdmtopModules:
\_module_inst: work@top (work@top), file:/home/tom/prog/git/Surelog/mytests/top.sv, line:1:1, endln:5:10
  |vpiName:work@top
  |vpiDefName:work@top
  |vpiTop:1
  |vpiTopModule:1
[...]
```

You should see that under `uhdmallModule` the tool has record my 3 modules : `A`, `top` and `B` none have `vpiName` attribute set,But instead you can get `vpiFullName` and `vpiDefName`

While if you look under `uhdmtopModule` the tool has only record a single module `top`, which is not instantiate; for which the `vpiName` has been set.

Depending of what you want to do and how you need to change a little bit the reference code to retrieve the data you are looking for.

dpetrisko, 2025-05-01:
Interesting! vpiFullName indeed works fine for me. This sounds like a bug in my understanding of the verilog object model rather than a parser bug, so I will close this issue. Thank you!

closed 2025-05-01 by dpetrisko