chipsalliance/UHDM #1105: SWIG question: uhdm_types e.g. vpiDesign

dpetrisko, 2025-05-05:
Hello, I'm trying to port uhdm-hier to the python wrapper, but there seem to be missing definitions from the SWIG export. I dug into it, but I'm unclear if this is intentional or not:

For instance, the type vpiDesign (defined as 3000) here: https://github.com/chipsalliance/UHDM/blob/a22bc0e48023ea931bbc482014c7c177a87c742d/templates/uhdm_vpi_user.h#L21

is checked in the user script:
https://github.com/chipsalliance/UHDM/blob/a22bc0e48023ea931bbc482014c7c177a87c742d/util/uhdm-hier.cpp#L85
```
if (vpi_get(vpiType, design) == vpiDesign) {
```

This should be included in the SWIG (from what I understand) here: https://github.com/chipsalliance/UHDM/blob/a22bc0e48023ea931bbc482014c7c177a87c742d/python/UHDM.i#L29

but it doesn't appear in the generated wrapper in contrast to something like vpiAlways:
https://github.com/chipsalliance/UHDM/blob/a22bc0e48023ea931bbc482014c7c177a87c742d/include/vpi_user.h#L123

```
$ uhdm.vpiAlways
1
$ uhdm.vpiDesign
AttributeError: module 'uhdm.uhdm' has no attribute vpiDesign
```

Are these "extra" types intentionally omitted? If so, how can we access them? If not, how can we fix it? Thanks!

Thomasb81, 2025-05-06:
Hi

vpi support allow some compatibility between tool. `vpiDesign` is not defined by ieee1800.  So I decided to avoid to include it in a first intention. Knowing that UHDM is open-source tool if someone need this symbol and is able to explain why, there is probably no good reason to stick to this position.
Feel free to to include missing symbol, if you need them.

dpetrisko, 2025-05-06:
Makes sense! I'm still confused why they are not included by the %include uhdm_types? Is it because they are nested or am I missing a ifndef SWIG somewhere?

Thomasb81, 2025-05-07:
It seems to me that `UHDM.i` does not include `uhdm_vpi_user.h` (`template/uhdm_vpi_user.h`)

dpetrisko, 2025-05-07:
It does include uhdm_types which itself includes uhdm_vpi_user.h: https://github.com/chipsalliance/UHDM/blob/a22bc0e48023ea931bbc482014c7c177a87c742d/templates/uhdm_types.h#L31

From my understanding SWIG is supposed to export defines from %include, but probably the nested include is what’s tripping it up. I’ll try including the types directly and seeing if that works

dpetrisko, 2025-05-07:
Including directly fixed it, thanks!

closed 2025-05-07 by dpetrisko